TheThingsNetwork/lorawan-stack #4095: Rate limit identical messages

johanstokking, 2021-04-28:
#### Summary

Rate limit identical messages coming from the same gateway.

#### Why do we need this?

Sometimes we see issues with the same message being forwarded many times by the same gateway.

This is often an issue in the gateway forwarder or concentrator, that concentrator registers aren't cleared correctly or that the forwarder software is simply in a loop.

#### What is already there? What do you see now?

We have rate limiting of UDP packets per session, and there is general rate limiting support in GS.

#### What is missing? What do you want to see?

Packet based rate limiting. This shouldn't be (only) in the UDP frontend. It should be per PHYPayload per gateway antenna, and it should expire after a short time. That is, in the order of tens of seconds.

#### How do you propose to implement this?

Since this is per gateway connection, there doesn't have to be persistency across multiple GS instances.

I would propose using the existing rate limiting infrastructure. The rate limiting key should contain the PHYPayload (plain, hash or CRC) and antenna index, and expire after 10 seconds. I think it's best to implement this in `(*GatewayServer).handleUpstream()` when reading from the `Up()` channel.

#### How do you propose to test this?

Unit testing

#### Can you do this yourself and submit a Pull Request?

Can review

neoaggelos, 2021-05-11:
If this is a known behaviour, and the problem is that gateways keep sending the same message over and over, then I suppose it is better to stick to a simpler solution of simply keeping track of the last message sent by the gateway and discard duplicates, similar to what [v2 is doing](https://github.com/TheThingsNetwork/gateway-connector-bridge/blob/master/middleware/deduplicate/deduplicate.go) ?

johanstokking, 2021-05-11:
Yes, that's probably the simplest solution.

Also note that in LoRaWAN, retransmissions are a feature. The same (un)confirmed uplink message can be sent multiple times, with the same FCnt, i.e. the same PHYPayload. Frequency hopping does apply though. Also, RX2 applies: the end device has to wait at least 2 seconds for the next frame.

So, if you want to work with a simple duplication detector on the last message, take into account either a) duplicate messages within 2 seconds or b) duplicate messages that are using the same frequency.

johanstokking, 2021-05-11:
Another related issue is that some gateways are sending duplicate messages of the same uplink frame:

1. Same PHYPayload
2. Different frequency
3. Different RSSI; e.g. -44 vs -100. The right one is the message with the highest RSSI
4. Out-of-order arrival of the messages; sometimes the one with the highest RSSI comes first, sometimes last
5. The frames are sent typically in the same `GatewayUp` message if the packet forwarder supports that (UDP, MQTT, but not Basic Station), so the timestamps are very close (no network latency)

This is likely caused by a bug in the SX1301, although I've only seen this with MultiTech Conduit and RAK (7258) gateways.

---

We can do two things:

1. Consider this a separate issue: assume that these duplicates are always in the same `GatewayUp` message, and simply loop over them and discard a dupe by RSSI
2. Extend the duplication filter (this issue) with a check by RSSI. Because of the out-of-order arrival of duplicate messages (the right one can be the second one) and the fact that multiple messages are processed one-by-one, this means that the filter needs to hold the messages for a short period of time (i.e. 10 ms [1]), see if there are dupes by PHYPayload and pass the one with the highest RSSI

@neoaggelos up to you which one you prefer implementing as part of this issue. If it's (1), file a separate issue for 3.13.1.

virtualguy, 2021-05-12:
I have seen similar duplicate messages from a multitech conduit but we have a 16 channel card configured in diversity mode, i.e. two SX1301 chips both listening on the same 8 channels. Is that a potential cause for the problem or is that unrelated?

johanstokking, 2021-05-12:
I don't think that is a potential cause, unless there's a race in the packet forwarder that mixes up PHYPayloads and the channel when reading the registers. If the issue is indeed with the SX1301, two of those increase the chance that it happens by a factor of two.

htdvisser, 2021-07-02:
This did not make it for the v3.13.3 deadline, so pushing to v3.14.0

PR #4395 (2021-07-14): For duplicate data uplinks, keep the one with highest RSSI

#### Summary

Closes #4095 (refs https://github.com/TheThingsNetwork/lorawan-stack/issues/4095#issuecomment-838443803)

#### Changes

- Introduce `io.DropDuplicatesByRSSI()`, which de-duplicates a list of gateway uplinks. When two or more uplinks contain the same payload, only the one with the highest RSSI is kept.
- Use it in gRPC and UDP frontends.

#### Testing

Unit tests

##### Regressions

Adds a few more memory allocations per uplink, but it should not be a problem.

#### Notes for Reviewers

`DropDuplicatesByRSSI` is a terribl [... cut]
closed 2021-07-14 by neoaggelos