TheThingsNetwork/lorawan-stack #4103: Gateways disconnected after recreate with different ID

cowjen01, 2021-04-29:
#### Summary

Hello, I had to change the gateway ID of my existing gateways, so I deleted them and then I recreated them with the same EUI. After that, both gateways are with in a disconnected status for 3 days. 

I check the gateway logs and traffic logs, all seems to be just fine. Before the recreation, both gateways worked without any troubles. I've also tried to reboot the gateway, but with no results.

When I try to list my gateways using `ttn-lw-cli g list` I see:

```
[{
  "ids": {
    "gateway_id": "iptec-3133303721006f00-gw",
    "eui": "3133303721006F00"
  },
  "created_at": "2021-04-26T15:03:14.792Z",
  "updated_at": "2021-04-27T11:32:15.235Z",
  "version_ids": {

  }
}, {
  "ids": {
    "gateway_id": "iptec-3133303754006500-gw",
    "eui": "3133303754006500"
  },
  "created_at": "2021-04-26T15:05:05.083Z",
  "updated_at": "2021-04-27T12:14:40.762Z",
  "version_ids": {

  }
}]
```

But when I try to check gateway's stats calling `ttn-lw-cli g get-connection-stats iptec-3133303754006500-gw` I get:

```
error:pkg/gatewayserver:not_connected (gateway `iptec-3133303721006f00-gw@ttn` not connected)
    gateway_uid=iptec-3133303721006f00-gw@ttn
    correlation_id=c581540c3ac24637b100f53a14793192
--- error:pkg/redis:not_found (entity not found)
```

I tried to delete one of them and wait, if there is not some kind of garbage collector, but from that moment I'm no longer able to create a new gateway with the changed ID.

```
{
  "code": 6,
  "message": "error:pkg/identityserver/store:id_taken (ID already taken)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/identityserver/store",
      "name": "id_taken",
      "message_format": "ID already taken",
      "correlation_id": "###",
      "code": 6
    }
  ],
  "request_details": {
    "url": "/users/cowjen01/gateways",
    "method": "post",
    "stack_component": "is"
  }
}
```

#### Steps to Reproduce

1. Create a gateway with some ID
2. Delete this gateway
3. Create the gateway with different ID but the same EUI

#### What do you see now?

Both of my gateways are disconnected.

#### What do you want to see instead?

They should be connected as before.

#### Environment

MacOS, Chrome, The Things Stack v3.12.1

nejraselimovic, 2021-04-30:
@cowjen01 thank you for reporting!

@KrishnaIyer do you know what might be an issue here?

KrishnaIyer, 2021-04-30:
> so I deleted them and then I recreated them with the same EUI.

So you recreated them with the same EUI but different ID?

> I tried to delete one of them and wait, if there is not some kind of garbage collector, but from that moment I'm no longer able to create a new gateway with the changed ID.

This is by design. We don't allow reusing IDs.

cowjen01, 2021-04-30:
@KrishnaIyer That's right! The previous ID was, for example, `iptec-testing-gateway,` and the new one is `iptec-3133303754006500-gw`, both with the same EUI. Moreover, I believe the previously created gateway is still active because I'm receiving new data through an application.

KrishnaIyer, 2021-04-30:
Yes so I think I see the issue. The Gateway Statistics Registry is not synchronised with the Entity Registry. So If a gateway is deleted when it's still connected, the connection stats for the gateway will continue to exist under the old ID until the gateway actually reconnects. This is by design since there isn't an easy way yet to force a server side disconnect.
- If you are using TCP based gateways (ex: LoRa Basics Station) then a simple restart/reconnection of the gateway will clear the stats of the old ID and you will have stats with a new ID.
- If you are using UDP gateways (it seems to me that this is the case), then there is no concept of a connection. The Things Stack monitors Status messages for the uplink path and `PULL_DATA` messages for the downlink path  and compares that with  the `DownlinkPathExpires` and `ConnectionExpires`  settings to check a gateway has lost connectivity. 

So in this case, your gateway must be turned off for at least as long as the `ConnectionExpires` interval (default 1 minute) for The Things Stack to ascertain that the gateway has lost connectivity and clear the stats for the old ID.

nejraselimovic, 2021-05-04:
> Yes so I think I see the issue. The Gateway Statistics Registry is not synchronised with the Entity Registry. So If a gateway is deleted when it's still connected, the connection stats for the [...]

@cowjen01 did this fix your issue?

cowjen01, 2021-05-04:
@nejraselimovic Sorry, I haven't got access to my gateway until now. I tried to turn the gateway off, wait for 1 minute, then turn it on, and it really helped solve the issue. Now I can see my gateway connected again.

I noticed there are some other people on the TTN forum who had the same issue. Maybe it would worth it to include this information in the documentation. Anyway, thank you very much for the help!

closed 2021-05-04 by cowjen01