TheThingsNetwork/lorawan-stack #4109: Building for development fails

jpmeijers, 2021-04-30:
#### Summary

Following the steps in https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/DEVELOPMENT.md there is an error when running the second command.

#### Steps to Reproduce

1. `git clone https://github.com/TheThingsNetwork/lorawan-stack.git`
2. `cd lorawan-stack/`
3. `make init`
4. `tools/bin/mage js:build`

#### What do you see now?

```
added 2884 packages, and audited 2886 packages in 4m

135 packages are looking for funding
  run `npm fund` for details

22 vulnerabilities (7 low, 11 moderate, 4 high)

To address issues that do not require attention, run:
  npm audit fix

To address all issues possible (including breaking changes), run:
  npm audit fix --force

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.
warning " > connected-react-router@6.5.2" has unmet peer dependency "react-router@^4.3.1 || ^5.0.0".                                                                                                      
warning "@storybook/addon-actions > @storybook/api@5.3.21" has unmet peer dependency "regenerator-runtime@*".                                                                                             
Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
(node:774697) Warning: Accessing non-existent property 'lineno' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
(node:774697) Warning: Accessing non-existent property 'column' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'filename' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'lineno' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'column' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'filename' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'lineno' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'column' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'filename' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'lineno' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'column' of module exports inside circular dependency
(node:774697) Warning: Accessing non-existent property 'filename' of module exports inside circular dependency
error Command failed with exit code 2.                                                                                                                                                                    
Error: running "npx yarn run webpack --config=config/webpack.config.babel.js" failed with exit code 2
```

#### What do you want to see instead?

The build should finish without an error.

#### Environment

* lorawan-stack commit ID c0e437afc7fccb9d506e05f73b4a2d2c3db54d13
* Ubuntu 20.04
* go version go1.16.3 linux/amd64
* npm v7.9.0

#### How do you propose to implement this?

...

#### How do you propose to test this?

Run the steps in DEVELOPMENT.md and see if any errors occur.

#### Can you do this yourself and submit a Pull Request?

No

jpmeijers, 2021-04-30:
```
$ npx browserslist@latest --update-db
undefined:1

SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at getLastestInfo (/home/user/go/src/lorawan-stack/node_modules/browserslist/update-db.js:57:15)
    at updateDB (/home/user/go/src/lorawan-stack/node_modules/browserslist/update-db.js:122:16)
    at Object.<anonymous> (/home/user/go/src/lorawan-stack/node_modules/browserslist/cli.js:40:3)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at internal/main/run_main_module.js:17:47
```

nejraselimovic, 2021-04-30:
Hi @jpmeijers,

can you please try running

> tools/bin/mage jsSDK:cleanDeps jsSDK:clean js:cleanDeps js:clean js:build

and let me know if this fixes your problem?

jpmeijers, 2021-04-30:
No, it's still failing.
[build-fail.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6405137/build-fail.txt)

nejraselimovic, 2021-04-30:
@kschiffer do you know what could be causing this? I've seen this error before but cleaning dependencies always worked this out for me.

jpmeijers, 2021-04-30:
I'm not sure if it is working yet or not, but I tried a couple of the suggested commands in the output:

* `npm audit fix --force`
* `npm i --package-lock-only`
* `npm audit fix`
* And again `tools/bin/mage jsSDK:cleanDeps jsSDK:clean js:cleanDeps js:clean js:build`

It looks like the error message is gone now, but I still see the same warnings on which it failed earlier. I'll try and run the rest of the stack now to confirm.

jpmeijers, 2021-04-30:
The stack looks like it is running fine.

nejraselimovic, 2021-05-04:
@jpmeijers cleaning dependencies is already documented in DEVELOPMENT.md as a general advice for when the build is failing, and it seems to help so I will close this issue for now.

closed 2021-05-04 by nejraselimovic