TheThingsNetwork/lorawan-stack #4112: `ns-db migrate` takes a long time for larger redis databases

antonysouthworth-halter, 2021-05-03:
#### Summary
Background: we use AWS Elasticache for Redis for ns-db. The "Cache Size" reported in the AWS Console is 804MB, and if we export that to `.rdb` format it's about 450MB.

Running `ns-db migrate` takes approximately 20 or so minutes, even when the database is already up to date (with `--log.level debug`, we are able to see all records skipped because they are already up to date).

This translates into an extra 20 minutes of downtime when the application is started, since we run the migrations on startup. This amount of downtime is not suitable for us as we have very strict uptime requirements.

For the meantime, we have changed our deployment such that we no longer run the migration on startup. However, it means we must now manually run the migrations, which is undesirable. Ideally, we wouldn't have to keep track of this manually.

#### Steps to Reproduce

1. Have a large Redis DB.
2. Run `ttn-lw-stack ns-db migrate`
3. Wait.....

#### What is already there? What do you see now?

Logs (edited slightly for formatting and redact IDs just to be safe):

```
+ ttn-lw-stack --log.level debug ns-db migrate
INFO Connecting to Redis database...         
DEBUG Skip valid key  key=ttn:v3:ns:devices:uid:REDACTED
DEBUG Skip valid key  key=ttn:v3:ns:devices:uid:REDACTED
DEBUG Skip valid key  key=ttn:v3:ns:devices:eui:REDACTED
DEBUG Skip valid key  key=ttn:v3:ns:devices:uid:REDACTED
DEBUG Skip valid key  key=ttn:v3:ns:devices:uid:REDACTED
DEBUG Skip valid key  key=ttn:v3:ns:devices:addr:REDACTED
# ... and so on, for ~10k or more keys
```

#### What is missing? What do you want to see?

`ns-db migrate` to be much faster, mainly in the case where all keys are already up-to-date. The faster the better (of course), but as a target, maybe <5 minutes?

#### Environment

Docker container `FROM thethingsnetwork/lorawan-stack:latest`.

#### How do you propose to implement this?

I notice [here](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.12/cmd/ttn-lw-stack/commands/ns_db.go#L123) that the migration seems to map a function over every record in the database.

I had a couple naiive thoughts about this; I'm sure someone who knows what they're doing will have better ideas:

- parallelise the operation over several cores w/ goroutines & GOMAXPROCS (same, but faster)
- record the schema version inside the keys and ask redis "select * where schema_version < $current_schema" (please excuse the pseudo-SQL, I don't have any experience querying Redis). (query fewer rows)
- record the state of the migrations inside the DB and only run if the recorded schema version < $current_schema
- re-order the `switch` statement so that the regex for checking valid keys runs first (atm it runs ~7 regex matches that fail before realising the key is already valid -- this would be a micro-optimisation but figure I mention it anyways).

#### Can you do this yourself and submit a Pull Request?

It would depend on my availability, and I would probably need a large amount of help from others.

KrishnaIyer, 2021-05-03:
ACK. We have seen this as well and I think it could do with some optimisation.

>  since we run the migrations on startup.

This is not necessary. Migrations are to be run only for minor updates and even then, only if the Changelog mentions an `ns-db` migrate. There is no need to run migrations at every startup if that's what you're doing here.

> However, it means we must now manually run the migrations, which is undesirable.

This is the intended mode of operation. These migrations by definition are to be run only when necessary.

kurtmc, 2021-05-04:
> This is not necessary. Migrations are to be run only for minor updates and even then, only if the Changelog mentions an ns-db migrate. There is no need to run migrations at every startup if that's what you're doing here.

> This is the intended mode of operation. These migrations by definition are to be run only when necessary.

Running database migrations before application startup is a really nice way to ensure that migrations have definitely been run. If you pick and choose when to run migrations in some other manual way you run the risk of forgetting and then deploying a completely broken application. In the case of lorawan-stack there's an extra level of complexity because for every version that is released, we as the consumer of your software need to very carefully read the release notes to check if there is some migration that needs to be run and then configure our deployment process to run that specific migration before application start. If you you guys ever forget to add to the release notes that we need to run a migration we might run into the same issue. On a previous occasion one of your releases required the configuration of `TTN_LW_CONSOLE_UI_EDTC_BASE_URL` which was not mentioned in release notes, this caused us to deploy the application is a degraded state, luckily only affecting the web page.

Running migrations before application startup is also a common pattern, it's done in a few of the open source tools that we run, one example (not related to LoRaWAN) is [concourse ci](https://github.com/concourse/concourse), the migrations are baked in and you don't even need to think about it when you upgrade version.

I guess it's not really my place to try influence the design decisions on this project, but I think it's important that I mention these issues to you guys otherwise nothing will ever change.

virtualguy, 2021-05-04:
I am also opposed to having to read the entire release note history for every upgrade. Its an error prone issue and seems so trival to solve. Is running a newer version without migrating first supported? I would expect that running a migration process to first detect the version of the data stored and then skip migrations if they were not required. Is there something fundamental I don't understand?

Its also particularly annoying that some components have a migration target and others do not. Surely we could future proof this by adding a migrate target which is stubbed out for each component. If this is of interest I am happy to look at putting up a PR

KrishnaIyer, 2021-05-05:
> I am also opposed to having to read the entire release note history for every upgrade. Its an error prone issue and seems so trival to solve.

The release history contains more information that just migration. We are internally looking at making the important information that require some operator action in every release more visible but that's documentation.
If you want to update to the new version of anything, you should read the release notes. This is not specific to TTS.

> I would expect that running a migration process to first detect the version of the data stored and then skip migrations if they were not required. Is there something fundamental I don't understand?

virtualguy, 2021-05-13:
Sounds good and I agree let's just focus on the original intent of the ticket, improving the migration performance

rvolosatovs, 2021-05-17:
I will see what can be done about this issue, but please do keep in mind that NS migrations are designed to be able to run at runtime, so you should be able to run `ns-db migrate` *after* the stack has already been started. That is why the implementation only considers one key at a time and everything is wrapped in transactions - to minimize the load on the database and ensure correctness even if the DB is being used.

nejraselimovic, 2021-06-07:
@neoaggelos please reassign to someone else if you cannot pick this one up

neoaggelos, 2021-07-07:
The discussion here has moved towards two separate directions:

- Skip running the ns-db migrations if they have already run, by recording the current schema version. This looks simple on paper as long the Network Server only uses Redis to persist data. This should be simple enough to do without any changes to `ttn-lw-stack`.

- Improve performance of the migrations themselves. I think this should be possible, since the migrations are operating on each device separately. I think parallelizing some of this work could indeed improve performance, at the cost of increased load on the Redis server.

So, what would the best way forward be for this issue? maybe @adriansmares can comment on this, also cc @johanstokking.

johanstokking, 2021-07-13:
> * Skip running the ns-db migrations if they have already run, by recording the current schema version. This looks simple on paper as long the Network Server only uses Redis to persist data. This should be simple enough to do without any changes to `ttn-lw-stack`.

I think this is quick and pretty safe to do. Do we just use a new Redis key with the TTS version of the last migration per component, that only gets set after a successful migration?

adriansmares, 2021-07-13:
> I think this is quick and pretty safe to do. Do we just use a new Redis key with the TTS version of the last migration per component, that only gets set after a successful migration?

I concur on this. No need for a set, map or anything like this, versioning will suffice.

PR #4410 (2021-07-20): Store schema version in ns-db migrate command

#### Summary

Closes #4112 

#### Changes

- Read schema version and only run migrations if needed 
- Record schema version after a successful migration run.
- Add `--force` flag to override this logic and always run migrations (if required)

#### Testing

Locally

#### Notes for Reviewers

`ns_db.go` already contains quite a lot of logic and raw Redis commands, so I went with a very simple implementation approach. 

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are [... cut]
closed 2021-07-20 by neoaggelos