TheThingsNetwork/lorawan-stack #4115: Sometimes missing 'decoded_payload' in Webhook JSON payload

phaabe, 2021-05-03:
#### Summary

The json payload which is forwarded via webhook integration sometimes misses `decoded_payload`.

#### Steps to Reproduce

Unfortunately, this is hard to reproduce...

Description of what happens sometimes:
In an application at TTN (things stack v3.12.1) almost all payloads are successfully decoded and the element `decoded_payload` appears in the forwarded Webhook JSON payload. In rare cases it is missing though.
One Webhook payload containing the raw payload `"frm_payload": "AQDHAiYDCSDNBw4kCwAAAAgNAA8AEgA="` missed the element `decoded_payload`. 
Trying to reproduce: When simulating a message on the same device and with the same port `AQDHAiYDCSDNBw4kCwAAAAgNAA8AEgA=` (hex `7e b9 a9 6b 29 68 69 d0 10 0c 70 22 60 30 92 0c d0 70 e2 40 b0 00 00 00 80 d0 00 f0 01 20 04`) is decoded successfully and appears in the Webhook JSON payload.

These findings make the reproduction of the error complicated.

#### What do you see now?

A webhook json payload which misses the `decoded_payload` even though the raw payload is clearly decodable.
...

#### What do you want to see instead?

A webhook json payload with the `decoded_payload`:
```json
"decoded_payload": 
{
 "temperature": 19.9,
 "humidity": 38,
 "x": 9,
 "y": 32,
 "z": -51,
 "vdd": 3620,
 "pulseAbs": 8,
 "digital": 0,
 "accMotion": 0,
 "waterleak": 0
}
```
...

#### Environment
The Things Network running the things stack v3.12.1.

Let me know if more information can help to analyze this. It looks like a bug. Somehow sometimes a payload is either not decoded or decoded but then not added to the JSON payload of the Webhook. Not sure if others have experienced this.

nejraselimovic, 2021-05-05:
@phaabe thank you for reporting! This is probably due to a large payload formatter script size that causes timeouts. See this issue: https://github.com/TheThingsNetwork/lorawan-stack/issues/4053

descartes, 2021-05-05:
I would suggest everyone consider the payload formatter as a bonus debug feature and do the decoding on their own application. It would be impossible to predict the perfect storm that means that even the simplest JS times out due to too many decoders to run - I am seeing it already with a 500byte decoder. If you are using a third-party dashboard, then live with the very occasional missing record.

closed 2021-05-07 by nejraselimovic