TheThingsNetwork/lorawan-stack #4130: How do we refer to the v3 Stack in third party offerings

descartes, 2021-05-05:
#### Summary
I have some sample API code ready to publish. What is the preferred way to reference the v3 Stack?

#### Why do we need this ?
Consistency in search terms mostly.

#### What is already there? What do you see now?
I am not aware of an official short name.

This is in the TheThingsNetwork GitHub account but is in reality being written by TheThingsIndustries

I currently call it TTNv3 but sometimes TTN v3 Stack but both assume a certain amount of assumed knowledge & domain knowledge, so perhaps LoRaWAN needs adding?

"The TTN LoRaWAN Stack v3 written by TTI staff"?

"TTI's LPWAN v3"?

#### What is missing? What do you want to see?

Your preference for a short hand way of referring to the new shiney.

#### How do you propose to document this?
I don't

#### Can you do this yourself and submit a Pull Request?
n/a

nejraselimovic, 2021-05-07:
@descartes An official name for the new Network Server is "The Things Stack". @laurensslats please tell if you prefer something more descriptive.

descartes, 2021-05-08:
Marketing & branding for something that you have to live with for many years to come, so please don't rush but if you do officially have an official name, it would be good to know.

kschiffer, 2021-05-21:
The official name is `The Things Stack`. Please avoid using `TTN v3` or any of its derivatives, as it only generates confusion.

If you want to refer to the deployment of `The Things Stack` that is running for The Things Network, you can refer to `The Things Stack Community Edition`. 

The namings of the other (commercial) offerings of TTS can be found here: https://www.thethingsindustries.com/deployment/

nejraselimovic, 2021-05-21:
@kschiffer thanks for getting back to this. @descartes You can see what Kevin described [here](https://www.thethingsindustries.com/docs/download/) too (though this page contains a small lapsus, it should be "The Things Stack Community Edition" instead of "The Things Network" for the free, community deployment, but the fix is about to be merged). I will close the issue as our official branding is completely in line with the aforementioned info.

closed 2021-05-21 by nejraselimovic