TheThingsNetwork/lorawan-stack #4143: Uplink simulator missing metadata which breaks payload formatters

benolayinka, 2021-05-06:
#### Summary

From https://www.disk91.com/2021/technology/lora/migration-to-ttn-v3/

The TTNv3 console allows simulating an Uplink. That way you can check the integration, even without devices communicating. Unfortunately, the message format is really simplified and you won’t be able to verify many things with this feature currently. As an example, the rx_metadata structure is not pushed over the uplink simulation. As a consequence you won’t be able to verify this critical part and you can also get some crash related to the missing part of the structure. So be careful when testing with this feature: I’ve spent 1 hour trying to fix crash when in fact the system was correctly working with normal device payload but not with simulated payload. At end my code is stronger but my time has been lost.

nejraselimovic, 2021-05-07:
An integration partner just told me that the fake gateway ID and frequency (plan?) are missing when simulating uplinks, which makes them harder to test integrations. @bafonins can we do anything about this?

descartes, 2021-05-21:
I too found this out the hard way - but I've not seen the fake data using the famous 42 format (kudos) missing. I'll look at my code in progress as I think there are some fields missing that you normally do get in a real uplink.

But regarding the situation of missing fields, it is totally reasonable for people to expect to program defensively - particularly as the protocol buffers can't cope with zero values so we never see things like f_cnt at 0.

bafonins, 2021-07-23:
As far as I can see, the Console pushes some test metadata together with uplink messages:
```json
{
  "uplink_message": {
    "f_port": ...,
    "frm_payload": "...",
    "rx_metadata": [
      {
        "gateway_ids": {
          "gateway_id": "test"
        },
        "rssi": 42,
        "channel_rssi": 42,
        "snr": 4.2
      }
    ],
    "settings": {
      "data_rate": {
        "lora": {
          "bandwidth": 125000,
          "spreading_factor": 7
        }
      }
    }
  }
}
```

@adriansmares Can you give more details on what we should include in the payload to resolve the issue?

bafonins, 2021-07-27:
As mentioned in my previous comment, the stack includes some default `rx_metadata` values when dispatching an uplink message. Please reopen this issue if any other fields should be added.

closed 2021-07-27 by bafonins