TheThingsNetwork/lorawan-stack #4145: ttn-lw-cli device list with more details

adrianimboden, 2021-05-06:
### Background
we have around 3500 devices in our application.
We use the cli to sync our list of devices and configuration (mainly appkey) with ttn. The process looks like this:
```
ttn-lw-cli device list app1
for each device in list:
  ttn-lw-cli device get app1 idXXX --root-keys.app-key.key
```

If the ```ttn-lw-cli device list``` command would support including more details (in our case root-keys.app-key.key), we could reduce the amount of api-calls from ~3500 to 4 (page size of 1000).

In TTN v2, this had been possible (the list already had the appkey included)

### Proposal
allow the following: ```ttn-lw-cli device list app1 --root-keys.app-key.key```

descartes, 2021-05-08:
As this data should be static, can you update the records for the newly added devices, not every single one?

Some tools I have been asked to develop have been spec'd to treat the stack database as the definitive source of data - but as you are finding, the APIs are not best suited to that and run the risk of overloading the server(s), particularly if you have more than one workstation running admin tasks, particularly if there is no form of caching involved.

I have a solution that I use for clients that holds the master copy of the data - additions are inserted in to TTN. Occasionally we run an extract of device ids and then royally kick the individual that entered it via the console without adding to the main database. Linked in to various time-series databases, we can do what we like, quickly, without impacting on the time sensitive servers.

I guess if there was to be any bulk data facility, a one-way replication of the Cockroach database to a setup that can take in various sorts of dump requests, queue them and send a notification with download link would be preferable.

If we build in convenience at the expense of server workload now, we will only impact on the scalability of any given set of hardware, a resource we need not add to because of a convenient architecture now.

adrianimboden, 2021-05-09:
Even if I don't want to thread the stack database as the definitive source of truth, it by design is that. The stack is that one that uses the data, therefore it is the source of truth.

The problem is, that I cannot guarantee that the data is not manipulated from anywhere else than our tool. We have the console, ttn-lw-cli users outside of the tool, different versions of the tool, etc., etc.

So the only 100% foolproof way to actually ensure that the data is in sync (which I do maybe once a year in the past or so), is to do a complete sync of the databases. And this currently needs in our case about 3500 API Calls which are unavoidable if I wan't to be absolutely sure that everything is allright.

descartes, 2021-05-09:
> We have the console, ttn-lw-cli users outside of the tool, different versions of the tool, etc., etc.

All of which can be restricted using the various permissions.

> ensure that the data is in sync (which I do maybe once a year in the past or so), is to do a complete sync of the databases

In that case, maybe that once a year impact rate limited out of region business hours is the best solution, rather than having people use the stack as a management database when it needs to be running a system critical application.

I'm raising this so that collectively we can come up with a good design that removes the potential of a burden that impacts on the running of the core functionality.

adrianimboden, 2021-05-11:
I also don't want the system to be burdened with useless calls. Thats the reason why I did come up with this issue after all.

I think data consistency/integrity is so important, that I want to verify that the systems are in sync. I would really appreciate a solution where this is possible in an end-to-end manner (which it is clearly not, when I'm not reading what is stored on the stack).

I mean, the exact data that I want to read (not write), is being used all 10mins because our devices send messages which have to be received and decoded on the stack. Why is it such a problem when I want to have the exact same information once a month or so?

I don't know the internal data structure, but when the data is directly adjacent to the ones which are sent anyways for the list view, this would benefit both sides without drawback I would assume.

nejraselimovic, 2021-05-12:
@neoaggelos can you please provide some input here?

neoaggelos, 2021-05-17:
Hi @adrianimboden,

Thank you for filing this issue. First, some background info: This stems from the fact that the data model of The Things Stack is vastly different from the The Things Network Stack V2.

The `AppKey` (as well as the `NwkKey` for LoRaWAN 1.1 devices) is only stored in the Network Server, and can even be encrypted at rest. The Identity Server does not (and _should_ not) know what the `AppKey` of the device is. The `ttn-lw-cli end-devices list` command maps to a `List` RPC to the Identity Server, so it is impossible to include the `AppKey` along with the result.

Further, even if it were to be implemented, the CLI would still have to do the 3.5K RPCs to the Network Server to fetch the required information.

> The problem is, that I cannot guarantee that the data is not manipulated from anywhere else than our tool. We have the console, ttn-lw-cli users outside of the tool, different versions of the tool, etc., etc.

It seems to me like the problem here is not to fetch every device from The Things Stack along with their `AppKey`, but rather to ensure that the `AppKey` cannot change. As @descartes mentioned, this should be static information.

If that is the end goal, The Things Stack has a rights system that can be used to great effect here. In particular, reading and/or changing the root keys of an end device implies that the user is a collaborator with the `APPLICATION_DEVICES_READ_KEYS` and/or the `APPLICATION_DEVICES_WRITE_KEYS` rights respectively.

You could make sure that "not-trusted" users do not get the `APPLICATION_DEVICES_WRITE_KEYS` right (maybe `APPLICATION_DEVICES_WRITE` also), which would solve your issue. You can then follow @descartes advice, and simply set the `AppKey` for newly-added devices only.

See also the [User Management](https://www.thethingsindustries.com/docs/getting-started/user-management/) and [Rights Management](https://www.thethingsindustries.com/docs/getting-started/user-management/rights/) pages from the documentation.

Hope this helps.

adrianimboden, 2021-05-17:
Thank you very much for the detailed explanation. Under these circumstances, my proposed changes makes no sense :+1:

closed 2021-05-17 by adrianimboden

StefanTzatz, 2022-03-01:
Hi, where is the console? I am new into TTN and IoT and by mistake I want to register again a gateway ID that i have already deleted without purge. As a result, I cannot use the same ID. I must delete it from console / CLI with this command in order to reuse it. I run the gateway with rasberry PI 4

descartes, 2022-03-02:
This is not a support channel - please use the forum ()https://www.thethingsnetwork.org/forum) or Slack for questions