TheThingsNetwork/lorawan-stack #4149: Accept and convert MAC address to Gateway EUI

htdvisser, 2021-05-10:
#### Summary

Some (DIY) gateways don't come with an EUI. For these gateways it could be nice if the Console would let users enter the MAC address of the gateway instead, and would then automatically fill the `FFFE` in the middle.

#### Background

> A MAC address can be turned into an EUI by inserting FFFE in the middle (e.g., `HH:HH:HH:HH:HH:HH` → `HH-HH-HH-FF-FE-HH-HH-HH`).

Source: https://doc.sm.tc/station/glossary.html?highlight=fffe#term-mac

#### What is already there? What do you see now?

Only a field for the EUI, and lots of questions on the Forum.

#### What is missing? What do you want to see?

Some assistance

#### How do you propose to implement this?

Perhaps we can add an event listener, that, when the EUI field loses focus, checks the length of the field, and when it's 12, would show a "convert this MAC address to an EUI" that users can click. When clicked, this would fill the `FFFE` in the middle.

#### How do you propose to test this?

E2E tests perhaps?

kschiffer, 2023-08-29:
Let's do as follows:

1. Add a debounced event listener to the change event
2. If the user stops entering the EUI after 12 characters (after debouncing for 3 seconds) OR blurs the field
3. Show a warning under the field (using the field warning) `It looks like you have entered a MAC address instead`
4. Convert the button next to the field to show `Convert MAC to EUI`
5. When clicking it, the EUI will be augmented with the `FF FE` value
6. The button will then return to its normal state and the EUI can be used to confirm the EUI as usual

Here's a small mockup:
[image]
[image]
[image]

KrishnaIyer, 2023-08-29:
This can be a good first issue right @kschiffer?

closed 2023-09-14 by adriansmares