TheThingsNetwork/lorawan-stack #4165: Render event data of type "google.protobuf.Value"

htdvisser, 2021-05-12:
#### Summary

It would be nice if the Console would automatically render event data with `"@type":"type.googleapis.com/google.protobuf.Value"` to the event list view as JSON.

#### What is already there? What do you see now?

In some places we add "simple" data to events, typically (lists of) strings or numbers. Such simple information is now missing from the event list, and can only be seen in the detail view.

One example is the `*.update` event that contains the list of updated fields.

```js
{
  "name": "gateway.update",
  // ...
  "identifiers": [
    {
      "gateway_ids": {
        "gateway_id": "..."
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/google.protobuf.Value",
    "value": [
      "attributes",
      "name"
    ]
  },
  // ...
}
```

Another example is the host that https://github.com/TheThingsNetwork/lorawan-stack/pull/4164 adds to the `gs.up.forward` event.

#### What is missing? What do you want to see?

Since it is often useful information, and since it's almost always relatively small, I'd like to see such event data in the event list, ideally rendered as JSON.

#### How do you propose to implement this?

> @bafonins: we already have setup to render events based on their `@type` entry. So it is just the matter of some UI/UX work and extending the list of supported data types for event entires _(sic)_ in the events view

---

It would be really nice if we could land this together with https://github.com/TheThingsNetwork/lorawan-stack/pull/4164, but if it's too much work, you can also select a different milestone.

kschiffer, 2021-05-12:
This is a simple matter of adding a preview for this data type in: [https://github.com/TheThingsNetwork/lorawan-stack/tree/v3.13/pkg/webui/console/components/events/previews](https://github.com/TheThingsNetwork/lorawan-stack/tree/v3.13/pkg/webui/console/components/events/previews)

and to add this new preview to the definitons in https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.13/pkg/webui/console/components/events/utils/definitions.js#L97-L108

htdvisser, 2021-05-18:
Closed by https://github.com/TheThingsNetwork/lorawan-stack/pull/4173

closed 2021-05-18 by htdvisser