TheThingsNetwork/lorawan-stack #4198: Add "last X minutes" parameter to Storage Integration API

moramos13, 2021-05-21:
#### Summary

On V2 Storage Integration API it was possible to add a "last X minutes" parameter on the API call.
...

#### Why do we need this?

This will be a good feature because currently the LNS sends all the stored uplinks to application, where are processed and the redundant data is erased. Also it waste network resources.
...

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?

...

#### Environment

...

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

...

descartes, 2021-05-21:
Whilst I can calculate the 'after' date, many of my more simplistic original scripts pull the data on a regular basis and just use a bit of overlap - so this would be useful. It also facilitiates a "I just want to see the last couple of hours uplinks for debugging purposes" type utility.

If this gets scheduled for a much later release, in the meanwhile I'll probably end up with some sample code to do this using the after field for Python, PHP & JavaScript anyway as I migrate my toolkit

nejraselimovic, 2021-05-21:
@johanstokking what do you think about this?

johanstokking, 2021-05-25:
I can imagine that this is very useful indeed.

@neoaggelos please triage.

closed 2021-07-01 by neoaggelos