TheThingsNetwork/lorawan-stack #4207: Refresh frame counters button

moramos13, 2021-05-25:
#### Summary

On V2 there was the possibility to refresh the frame counters of a given device and it was very useful when programming the end-device. Currently, I can't find this option on the application interface, so everytime a chance must be done on the end-device program, a new device should be created on the LNS.

#### Why do we need this?

For debbuging purposes.

#### What is already there? What do you see now?

The interface shows the uplink and downlink counters.

#### What is missing? What do you want to see?

Button that refresh the frame counters.

#### Environment

Win10/Brave Browser/RAK2245/MKRWAN1310

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

...

nejraselimovic, 2021-05-28:
@johanstokking resetting frame counters on The Things Stack is only available for ABP devices and not for OTAA, correct? Could you comment on this proposal?

kschiffer, 2021-06-01:
Duplicate of #2434

closed 2021-06-01 by kschiffer