TheThingsNetwork/lorawan-stack #4215: Deletion of multiple end devices in Console

willhelmx, 2021-05-28:
_Summary:_
Possibility to delete several end devices in a row under 1 application in Console

_Why do we need this?_
For easy handling of multiple end devices under 1 application

What is already there? What do you see now?
Today we cannot delete an entire apllication without deleting the devices, the devices on the other hand, can only be deleted one by one..

#### What is missing? What do you want to see?
A way to delete multiple devices at the time.

#### How do you propose to implement this?

As we see it to prevent accidental deletion one of two ways can be implemented
1) In order to delete all -> set token to be entered before execution
and/or
2) Have checkboxes for every device that is to be affected.

nejraselimovic, 2021-06-02:
@htdvisser I see that IS refuses to delete an application if it still contains some registered devices, but this suggestion looks useful.  Is there any argument against doing this?

htdvisser, 2021-06-02:
I like the idea of having checkboxes to select multiple devices that need to be deleted.

This could of course get a bit tricky in applications that have devices registered in different clusters, because each cluster Console is only supposed to manage devices in its own cluster (at least, until we have a global Console). So if we let users delete multiple devices using checkboxes, the Console should somehow inform the user which (if any) devices need to be deleted in a different cluster Console.

johanstokking, 2021-07-20:
We need thumbs ups on this issue before we pick this up. I see the need for it, but we need a bit more demand for this before we plan this.

willhelmx, 2021-07-20:
Hi all... i beleive that the demand for this functions will grow along with the business growth within the system so eventually it needs to be implemented.
If we focus on just small and private enterprises there is perhaps no need for it, but if you focus on setting up 100´s of end nodes in a row it will be a total nightmare if something goes wrong.

johanstokking, 2021-07-20:
@willhelmx I understand, we'll monitor the thumbs up on this comment.

Large scale deployments can already use the API and scripts using the CLI.

sanijs, 2022-01-04:
+1 for this.

Is it correctly understood that if I want to delete multiple end devices using the API I need to do it 4 times for each end device?

```
DELETE /api/v3/as/applications/{application_ids.application_id}/devices/{device_id}
DELETE /api/v3/ns/applications/{application_ids.application_id}/devices/{device_id}
DELETE /api/v3/js/applications/{application_ids.application_id}/devices/{device_id}
DELETE /api/v3/applications/{application_ids.application_id}/devices/{device_id}
```

willhelmx, 2022-01-04:
Hi again Guy´s

Hopefully, there is a solution for this on the way
The most intuative would be to have checkboxes on the left of the console device list in order to select specific item or all for deletion..

Have a great day ladies and gentlemen

Best Regards
/ Anders

htdvisser, 2022-01-04:
@sanijs: that's right (and in case you're wondering, that's something we want to make less complicated in a future V4 API).

Also: instead of commenting "+1", please put a 👍 reaction on the issue, so that we can track demand and prioritize this accordingly.

@willhelmx: this is currently not planned. We have limited resources and other bugs/features that have higher priority or more demand. As @johanstokking already commented, you can use the API or script something with CLI.

KrishnaIyer, 2024-10-10:
This is tracked [internally](https://github.com/TheThingsIndustries/lorawan-stack/issues/3736) and is planned for 2024 Q4.

KrishnaIyer, 2024-10-22:
Replaced by https://github.com/TheThingsNetwork/lorawan-stack/issues/7339

closed 2024-10-22 by KrishnaIyer