TheThingsNetwork/lorawan-stack #4218: TTNv3 : Add gateway location to the metadata of uplink messages

meten-natuurlijk, 2021-05-28:
#### Summary
In TTNv2 an MQTT JSON uplink message contains the location of gateways in the metadata.
This metadata seems to be missing in the TTNv3 MQTT JSON metadata.

#### Why do we need this?
We receive financial donations from our local municipality to place gateways and create TTN coverage in our area.
We justify these investments by reporting the location, availability and coverage of gateways in a real-time map.
We also use the gateway metadata to actively monitor the availability of the gateways.

#### What is already there? What do you see now?
There is a lot of (broker) metadata in the TTNv3 uplink message, but the location of gateways is missing.
From the perspective of an TTN end-user we believe that the location of gateways has more value than the broker (timing) metadata.

#### What is missing? What do you want to see?
We would like to see the location of gateways in the metadata of an uplink message.

#### Environment
* The console of the Things Stack - Community edition
* The JSON payload of MQTT TTNv3 uplink messages

#### How do you propose to implement this?
There are multiple scenario's to fix this:
1) Add the location of gateways to the metadata. The size of the metadata would increase.
2) Add the location of gateways, and remove some broker metadata. The size of the metadata would effectively decrease.
3) Add an option to the TTNv3 console. This would allow an application-maintainer to enable/disable the adding of gateway locations to the JSON metadata. This way the location of gateways is only added to the metadata for applications that actually need it.

#### How do you propose to test this?
We can help testing the MQTT JSON payload.

#### Can you do this yourself and submit a Pull Request?
Sorry, we don't have the technical skills to produce a solution/fix and create a pull-request.

nejraselimovic, 2021-06-02:
@johanstokking @KrishnaIyer @htdvisser We discussed this issue [before](https://discuss.thethingsindustries.com/t/can-we-learn-something-from-community/483/5). Any thoughts? Can we decide on the approach to solving this issue, so we can schedule it for implementation?

htdvisser, 2021-06-02:
@neoaggelos wasn't this metadata already supposed to be there? As far as I remember, if gateways have their location set to public, the locations should also be included in uplinks, right?

neoaggelos, 2021-06-03:
This metadata is indeed included already. The important note is that the **Publish Location** box must be ticked in the **Location** settings pane. Make sure to restart the gateway for the changes to take effect.

[image]

The linked PR (https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/375) adds an example of location metadata in the [Data Formats](https://www.thethingsindustries.com/docs/reference/data-formats/#uplink-messages) page in the documentation.

closed 2021-06-04 by nejraselimovic