TheThingsNetwork/lorawan-stack #4219: Console uplink data: show data rate in stead of bandwidth

kersing, 2021-05-28:
#### Summary
Please show the data rate in the live data summary of uplinks in the console. The data rate implies the bandwidth (because it is defined in the regional parameters) and additionally indicates the spreading factor.

#### Why do we need this?
Currently the only way to view the data rate and/or spreading factor is to open an uplink packet and search it in the huge amount of data shown. Reading and searching takes time when this is basic information regarding a nodes performance that should be available at a glance.

#### What is already there? What do you see now?
Currently the bandwidth is listed which is not very useful as it is (at least for EU868) fixed anyway.
Uplink line:
---
DevAddr 26014811 FCnt 10654 FPort   2 MAC payload 4FF554BE6C7B69 Bandwidth 125000 SNR 9.5 RSSI -85 Raw payload...
---

#### What is missing? What do you want to see?
The data rate.

#### Environment
N.a.

#### How do you propose to implement this?
Replace bandwidth with data rate.

#### How do you propose to test this?
Look at live data view of a gateway or an application/device

#### Can you do this yourself and submit a Pull Request?
Probably but I suspect the UI designers do it in <10% of the time it takes me.

nejraselimovic, 2021-06-02:
@kersing this does seem useful! @kschiffer can you pick this up?

htdvisser, 2021-06-22:
I think what's most useful for users is to combine the `bandwidth` and `spreading_factor` to a single string such as `SF7BW125`.

PR #4372 (2021-07-12): Improve uplink data event preview
#### Summary
Closes #4219 

#### Changes
- Display data rate (with both bandwidth and spreading factor) in `UplinkMessage` and `DownlinkMessage` event preview
- Remove `MAC payload` and `Raw Payload` previews
- Remove Raw payload and MAC payload from `UplinkMessage` event
- Improve JSON preview of arrays within decoded payloads (show preview instead of ellipsis, when array consists only of primitives)
- Mark confirmed uplinks and downlinks in event previews
- Use a UI switch instead of checkbox for controlling stream verbosity

[image]

#### Testing

Manual

# [... cut]
closed 2021-07-12 by kschiffer