TheThingsNetwork/lorawan-stack #4228: Basics Station - Wrong uri returned by /router-info when using unencrypted traffic

ogimenez-smtc, 2021-06-01:
#### Summary
The Things Stacks returns an secured websocket address instead of unsecured when using an unencrypted traffic with Basics Station

#### Steps to Reproduce

1. Set `tc.uri` to an unsecured endpoint, ex: `ws://eu1.cloud.thethings.network:1887`
2. Do not set `tc.trust` file
3. Set correct API key in `tc.key`
3. Start Basics Station

#### What do you see now?

Basics Station log:
```
2021-06-01 12:58:01.104 [TCE:INFO] Connecting to INFOS: ws://eu1.cloud.thethings.network:1887
2021-06-01 12:58:01.130 [AIO:XDEB] [3] ws_connecting state=3
2021-06-01 12:58:01.130 [AIO:XDEB] [3|WS] > {"router":"0:ff:fe00:0"}
2021-06-01 12:58:01.163 [AIO:XDEB] [3|WS] < {"router":"ff:feXX:XXXX","muxs":"muxs-::0","uri":"wss://eu1.cloud.thethings.network:1887/traffic/eui-000000FFFEXXXXXX"}
2021-06-01 12:58:01.163 [TCE:INFO] Infos: ff:fe00:0 muxs-::0 wss://eu1.cloud.thethings.network:1887/traffic/eui-000000FFFEXXXXXX
```

returned URI is `wss://` instead of `ws://`. Port is correct

#### Can you do this yourself and submit a Pull Request?

Unfortunately no

KrishnaIyer, 2021-06-01:
Technically we shouldn't allow plaintext port. I'll check with ops to disable this.

ogimenez-smtc, 2021-06-01:
Thanks, I am not sure to understand if you want to disable it for TTN/TTI hosted instances or in the code itself.

I think that it should be disabled in the code because this issue also applies to self hosted instances

KrishnaIyer, 2021-06-01:
> Thanks, I am not sure to understand if you want to disable it for TTN/TTI hosted instances or in the code itself.
> I think that it should be disabled in the code because this issue also applies to self hosted instances

This is for hosted instances. 

For self hosted, this can be disabled using the `gs.basic-station.use-traffic-tls-address` option.

ogimenez-smtc, 2021-06-01:
Ok, I had to dig in the code to understand that you set this option in hosted instances making [this test](https://github.com/TheThingsNetwork/lorawan-stack/blob/c0e437afc7fccb9d506e05f73b4a2d2c3db54d13/pkg/gatewayserver/io/ws/ws.go#L120) `True`, so it forces wss scheme in the returned URI, thanks.

In such case I would expect the returned URI port to be `8887` instead of `1887` as we cannot establish secure connection on 1887

KrishnaIyer, 2021-06-01:
> In such case I would expect the returned URI port to be 8887 instead of 1887 as we cannot establish secure connection on 1887

When deployed behind a proxy, TTS cannot know what the port or if it's plain text or TLS. That's why this setting exists.

ogimenez-smtc, 2021-06-01:
ok, thank you

KrishnaIyer, 2021-06-08:
So I did some checking and this is actually intended. Since we enforce auth check for the `/traffic` endpoint, this should return a WSS link. So this works as intended imo.

closed 2021-06-08 by KrishnaIyer

ogimenez-smtc, 2021-06-08:
Ok thanks