TheThingsNetwork/lorawan-stack #4267: LoRa Cloud user experience and minor functional issues

johanstokking, 2021-06-14:
#### Summary

LoRa Cloud user experience and minor functional issues

Documentation improvements are already tracked in https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/266

#### Steps to Reproduce

This spans LoRa Cloud Device and Application Services (DAS) and Geolocation integration. Setup is standard, see for instance https://www.thethingsindustries.com/docs/integrations/cloud-integrations/ubidots/lora-edge-geolocation/#prepare-a-lora-cloud-setup as reference.

#### What do you see now?

1. **Console**: When first expanding the DAS token view and afterwards the GLS token view in the integration tab, then the DAS token is gone. The Geolocation token has a similar issue
2. The "automatic multi-frame" mode (looking at first byte, use value as `N` with maximum of 16) isn't configurable. Is this implemented at all? Is this always enabled?
3. Device location isn't updated in the Identity Server (or not visible in the Console)

#### What do you want to see instead?

1. The tokens should probably stay in the view state
2. Should the automatic multi-frame be an explicit setting? Or does it simply need documentation?
3. Successful geolocation responses should result in the device moving on the map in the Console

Other requested features:

A. Include the request body in the service data. This could be useful to debug geolocation output, by knowing the input of the request. This may also become useful with other integrations.

#### How do you propose to implement this?

1. This should just be a frontend thing; we don't need new API support for this. If people previously configured a token but did not enable the integration, the token is gone. It's just that when staying on the screen and disabling and re-enabling things, the tokens should come back
2. Not sure
3. The location of the device should be updated in IS with cluster auth

A. Consider including the request body always in the service data event type

#### How do you propose to test this?

Integration testing

#### Can you do this yourself and submit a Pull Request?

Can test and review

adriansmares, 2021-06-23:
> Should we make this more explicit, with a checkbox?
There is a small text below it that explains what setting it to 0 does, but I can agree that a checkbox would help - I just wasn't able to implement this nicely - maybe @kschiffer or @bafonins could pick this one up.

> Include the request body in the service data. This could be useful to debug geolocation output, by knowing the input of the request. This may also become useful with other integrations.

The message itself is very big as is, so I'm not really happy to add this. In my opinion it only helps debugging. 
I would instead like to see a debug level message with the body instead.

> The location of the device should be updated in IS with cluster auth

My idea here was to have a package that does this - listens for `ttnpb.ApplicationUp_LocationSolved` and updates the location of the end device. Then we don't need to do this multiple times (i.e. for DAS too). Another useful thing would be to have multiple tabs in the Console with the location from each integration - keep in mind that end device location is a dictionary between the name of the source and the locatino itself.

johanstokking, 2021-06-24:
> My idea here was to have a local listener (like `webhook` or `applicationpackages`) that listens for `ttnpb.ApplicationUp_LocationSolved` and updates the location of the end device. Then we don't need to do this multiple times (i.e. for DAS too).

Can't it be here? https://github.com/TheThingsNetwork/lorawan-stack/blob/e28b20e36be1ae222ef9853d77f7b8fd815ef537/pkg/applicationserver/applicationserver.go#L685

adriansmares, 2021-07-02:
The functional issues with WiFi / GNSS should be fixed now, leaving the Console fixups to @kschiffer (please reassign if needed).

kschiffer, 2021-07-16:
> 1.  **Console**: When first expanding the DAS token view and afterwards the GLS token view in the integration tab, then the DAS token is gone. The Geolocation token has a similar issue

I'm not able to reproduce. Has this been fixed in the meantime?

> 1.  Should we make this more explicit, with a checkbox?

I need some more info on this. I see that there is already a checkbox for enabling Multiframe. What else would be needed here?

adriansmares, 2021-07-16:
> I'm not able to reproduce. Has this been fixed in the meantime?

I didn't touch it, but I haven't been able to reproduce it either.

> I need some more info on this. I see that there is already a checkbox for enabling Multiframe. What else would be needed here?

The checkbox is about the window size - the idea is that a window size of 0 means that the AS decides the window size automatically.
In order for a user to understand that this happens, he needs to read the description of the window size. This is a big if.
It would be a bit more intuitive for him to have a chebkox under multiframe called 'automatic window size', which when unchecked reveals the actual form input that allows him to input a value over one. When the checkbox is checked, the window size should be 0.

PR #4416 (2021-07-20): LoRaCloud form fixes
#### Summary

This PR addresses the outstanding Console issues of #4267. 

[image]

Closes #4267 
Closes #4099

#### Changes
- Move fetching from container to view

#### Testing

- Manual testing

##### Regressions

This could introduce issues to the LoRaCloud DAS forms in the Console.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch [... cut]
closed 2021-07-20 by kschiffer