TheThingsNetwork/lorawan-stack #4268: Selecting text from Live Data Event Details unclear highlighting

jpmeijers, 2021-06-14:
#### Summary

When highlighting text to select and copy from the event details tab under the live data feed, it is unclear which characters are selected.

Example:
[image]

The above selected highlighting starts in the middle of the first `"` and ends in the middle of the last `"`. It is unclear if the first and last `"` are included in the selection or not.

When I copy in the above example, what is pasted is:
`ACABAAAApQ=="`

#### Steps to Reproduce

1. Select text and try to guess what is selected
2. Paste text and see if it matches
3. ...

#### What do you see now?

Highlighting starts and ends on characters, not between them as expected.

#### What do you want to see instead?

Make highlighting of text behave as in normal text editors.

#### Environment

Google Chrome Version 91.0.4472.77 (Official Build) (64-bit)
Ubuntu 20.04.2
TTS 3.11.1

#### How do you propose to implement this?

Maybe change the font that is used.

#### How do you propose to test this?

Select text and see if the highlighting snaps to between characters, not on them.

#### Can you do this yourself and submit a Pull Request?

No

kschiffer, 2021-06-15:
Related to #4195

I can confirm this issue and it seems to happen only for some lines in the Code Editor. We will hopefully fix this by replacing the component which has a number of other issues as well.

johanstokking, 2021-07-13:
Let's track this in #4195 and reopen or open a new issue if the problem persists.

closed 2021-07-13 by johanstokking