TheThingsNetwork/lorawan-stack #4271: Console timestamps are truncated rather than rounded

V999TEC, 2021-06-15:
#### Summary
This is a very minor issue but I noticed the timestamps are up to a second out due to truncation.
By using rounding it need only be out by up 0.5 sec. 
...

#### Steps to Reproduce

1. Select  any gateway/name/Live Data to show a list of Time/Type/Data preview
2. The Time shown is typically local time HH:MM:SS
3. Now select any event detail such as "gs.status.receive" and look at the precise "time": "value" in the json
4. The value of time is precise and accurate UTC
5. Comparing the precise UTC to the local time it is clear that the HH:MM:SS is not rounding the seconds.

#### What do you see now?
Example (I am UTC+1) 
10:15:45

{
  "name": "gs.status.receive",
  "time": "2021-06-15T09:15:45.982495171Z",
  "identifiers": [
...

#### What do you want to see instead?

Would be better rounded as
10:15:46

...

#### Environment
Chrome browser Version 91.0.4472.101 (Official Build) (64-bit)

...

#### How do you propose to implement this?
This is likely to be a 1 line fix/improvement

...

#### How do you propose to test this?
Very easy to verify by comparing timestamps where seconds <0.5, >=0.5
And better still when that increments MM amd/or HH
i.e., 09:59:59.9954321
rendered in the summary list as 10:00:00 rather than 09:59:59
(I'm assuming in this example that my local time is UTC+0)
...

#### Can you do this yourself and submit a Pull Request?
Sorry no

...

nejraselimovic, 2021-06-16:
@kschiffer can you please explain why do we have such behavior (rather than the one proposed)?

V999TEC, 2021-06-16:
By using trucation of the precise timestamp it couple be up to a second in error.
If you round the decimal places of the seconds down if <0.5 and up if >= 0.5 then the error is reduced by half.

But this is only a personal preference: the current behaviour is not wrong it's just more common to see rounding done in this sort of scenario.  
Say I have a couple of messages, one at 42.1 seconds and another at 42.9 the current trucation makes it look like both occured at 42. Far better in my opinion if the first showed 42 and the second 43, since the time error is 0.1 for both 1st and 2nd message. In contrast, the truncation implies an error of 0.1 + 0.9 for 1st and 2nd messages respectively.

kschiffer, 2021-06-17:
I guess we could also round the time values (even though I would find it a bit weird and don't think it is quite common) but I think a better solution is increasing the resolution of the time display in the event view to milliseconds. I can look into that as part of #3817.

johanstokking, 2021-07-13:
Times should be truncated; it should not be rounded up because that is incorrect. The event happened in the second it is displayed, not in the second after. I added a small note in #3817 to add some fraction. Closing this issue. Thanks for reporting and sharing your opinion though!

closed 2021-07-13 by johanstokking