TheThingsNetwork/lorawan-stack #4285: Add note/tooltip in the console when adding payload formatters

rish-c, 2021-06-18:
#### Summary
With the new size limit for payload formatters, users are not able to create new applications using existing formatters. They should be prompted to use the Device Repository instead. 

#### Why do we need this ?
It is not clear what to do when they exceed the limit

#### What is missing? What do you want to see?
A small note or a tool-tip when adding payload formatters to use the DR.

#### How do you propose to document this?
N/A

#### Can you do this yourself and submit a Pull Request?
Need @kschiffer

descartes, 2021-06-21:
Some forum users have suggested that having a simple UI to access the repository data would be useful to help them adjust their migrated devices and have access to the payload formatter as copying it over from v2 isn't feasible (apparently).

I've rescripted some of my CLI migration helpers to pick up the device repository links, so one possiblility is an migration tool command line option to inject the required fields for a device in to the output.

kschiffer, 2021-07-13:
I'll close this since:

- I don't think such tooltip is the proper solution. Users with custom (big) payload formatters will not be helped by using the PF of the DR
- The size limit for PFs has been updated recently via #4133 
- I'll open another issue about showing the DR PF script in the console

closed 2021-07-13 by kschiffer