TheThingsNetwork/lorawan-stack #4288: Payload Formatter JavaScript rounding

descartes, 2021-06-21:
#### Summary
The update to 3.13.2 has introduced some rounding issues in the payload formatter processing

#### Steps to Reproduce

Payload: 00270611321932135a18830e1918a80ff31925

var b = input.bytes; var d = {};
d.Flags = b[0];
d.MCUTemp = ((b[1] << 8) + b[2]) / 100;
var l = (b.length - 3) / 4;
for (var c = 0; c < l; c++) { 
	d["Moisture" + (c+1)] = (((b[c*4+3] << 8) + b[c*4+4]) / 100) - 10;
	d["Temperature" + (c+1)] = (((b[c*4+5] << 8) + b[c*4+6]) / 100 - 50);
}

#### What do you see now?
"decoded_payload": {
        "A": "Data",
        "BFTxt": "",
        "BFlags": 0,
        "MCUTemp": 99.9,
        "Moisture1": 34.02,
        "Moisture2": 39.54,
        "Moisture3": 26.090000000000003,
        "Moisture4": 30.83,
        "Temperature1": 14.5,
        "Temperature2": 12.75,
        "Temperature3": 13.119999999999997,
        "Temperature4": 14.370000000000005
}

#### What do you want to see instead?
"decoded_payload": {
        "A": "Data",
        "BFTxt": "",
        "BFlags": 0,
        "MCUTemp": 99.9,
        "Moisture1": 34.02,
        "Moisture2": 39.54,
        "Moisture3": 26.09,
        "Moisture4": 30.83,
        "Temperature1": 14.5,
        "Temperature2": 12.75,
        "Temperature3": 13.12,
        "Temperature4": 14.37
}

The firmware reports everything OK, prior to 3.13.3 it was OK, so the JS interpreter must have had some form of update ...

#### Environment
TTS CE

#### How do you propose to implement this?
Not me

#### How do you propose to test this?
Can check output when it's done

#### Can you do this yourself and submit a Pull Request?
Nope

#### Additional ...
I can put a toFixed(2) in to the code - except I'm trying to keep my formatter small ...

johanstokking, 2021-06-22:
> The firmware reports everything OK, prior to 3.13.3 it was OK, the figures are correct subject to rounding issues, so the JS interpreter must have had some form of update ...

Yes, that might be. And you don't want to hear this, but we have no control over how this interpreter is handling floating point. Not to mention that rounding is a known problem with floating point. Let alone in JavaScript.

You might want to subtract the offset before dividing by 100, so `((...) - 5000) / 100`.

> Steps to Reproduce

What is `c4`?

```
d["Moisture" + (c+1)] = (((b[c4+3] << 8) + b[c4+4]) / 100) - 10;
                             ^
ReferenceError: c4 is not defined
```

descartes, 2021-06-28:
> And you don't want to hear this, but we have no control over how this interpreter is handling floating point. Not to mention that rounding is a known problem with floating point. Let alone in JavaScript.

Yeah, got several T Shirts on that - just a bit odd that it changed and as above, adding in code means we increase the formatter size. No worries, bit safer being explicit with the toFixed().

All that said, I only use PF's for testing / development - deployments decode on client servers.

> What is `c4`?

A typo, should have been c*4, no idea how that occured.

Happy for this to be closed if the JS interpreter is a blackbox.

johanstokking, 2021-06-28:
```
$ node -v
v16.3.0
$ node index.js
{
  Flags: 0,
  MCUTemp: 99.9,
  Moisture1: 34.02,
  Temperature1: 14.5,
  Moisture2: 39.54,
  Temperature2: 12.75,
  Moisture3: 26.090000000000003,
  Temperature3: 13.119999999999997,
  Moisture4: 30.83,
  Temperature4: 14.370000000000005
}
```

johanstokking, 2021-06-28:
Note that `toFixed(2)` returns a string, which is really good for display but probably not what you want as the output of codecs.

---

My previous suggestion works BTW;

```js
var b = input.bytes; var d = {};
d.Flags = b[0];
d.MCUTemp = ((b[1] << 8) + b[2]) / 100;
var l = (b.length - 3) / 4;
for (var c = 0; c < l; c++) {
d["Moisture" + (c+1)] = (((b[c*4+3] << 8) + b[c*4+4] - 1000) / 100);
d["Temperature" + (c+1)] = (((b[c*4+5] << 8) + b[c*4+6] - 5000) / 100);
}
```

Yields:

```json
{
  Flags: 0,
  MCUTemp: 99.9,
  Moisture1: 34.02,
  Temperature1: 14.5,
  Moisture2: 39.54,
  Temperature2: 12.75,
  Moisture3: 26.09,
  Temperature3: 13.12,
  Moisture4: 30.83,
  Temperature4: 14.37
}
```

You got the t-shirts so I guess you know why 😉

nejraselimovic, 2021-06-29:
@johanstokking why was this reopened? Is there something that needs to be done here?

johanstokking, 2021-06-29:
I accidentally closed it and reopened it, but I believe it can be closed now. This can be "fixed" in the script. We can't promise anything about floating points digits I'm afraid.

closed 2021-06-29 by johanstokking

descartes, 2021-07-01:
> Note that `toFixed(2)` returns a string, which is really good for display but probably not what you want as the output of codecs.

Indeed, but it calmed the client down who was screaming in my ear whilst I was doing it! And I use too many languages, too many T Shirts, today's is for something called Symantec Café, one of the early Java IDE's, so at least 28 years old ...