TheThingsNetwork/lorawan-stack #4290: Change some default values in the Console

johanstokking, 2021-06-21:
#### Summary

Change some default values in the Console

#### Why do we need this?

To make sure that we have saner defaults when people don't change the values.

#### What is already there? What do you see now?

1. RX1 delay defaults to 1 second when not entered. This happens at least when creating ABP devices
2. Gateway status is not public by default. There is no explanation what this means
3. Gateway location is not public by default. There is no explanation what this means

#### What is missing? What do you want to see?

1. Default RX1 delay should be 5 seconds, and this should also be displayed in the input box
2. Gateway status should be public by default. There should also be a short description that only if the location is public, the gateway status is shown on maps. Mapping services (Packet Broker, TTN Mapper) may have their own access policies (not sure if we need to mention the mapping services by name or just explain the concept)
3. Gateway location should be public by default. See (2) for explanation

#### How do you propose to implement this?

This should probably be baked in the Console, but I can also imagine that we're missing a default value in the NS (?). Also check how CLI handles this

#### How do you propose to test this?

Local

#### Can you do this yourself and submit a Pull Request?

No

htdvisser, 2021-06-24:
(2) and (3) are done with #4296

johanstokking, 2021-06-24:
Great! Are assignment, labels and milestone up to date here?

htdvisser, 2021-06-24:
Yeah, I'll see if I can still do (1) before the v3.13.3 deadline, otherwise I'll reschedule or re-assign.

htdvisser, 2021-06-28:
@johanstokking I don't think we should change that RX1 delay from 1 to 5 seconds just like that.

Instead, I would like to propose that we create a toggle between "Device starts with LoRaWAN default settings" and "Device starts with frequency plan settings", and then fill those into the following fields:

- RX1 Delay
- RX1 DR Offset
- RX2 DR Index
- RX2 Frequency
- Factory Preset Frequencies

What do you think?

cc: @kschiffer @bafonins

johanstokking, 2021-06-28:
> "Device starts with frequency plan settings", and then fill those into the following fields:
> 
> * RX1 Delay
> * RX1 DR Offset
> * RX2 DR Index
> * RX2 Frequency
> * Factory Preset Frequencies

RX1 delay isn't part of the frequency plan.

The default and desired RX1 delay is determined as follows:

https://github.com/TheThingsNetwork/lorawan-stack/blob/4466080950116b2cc50cecdaef57f9ce0da3a9d3/pkg/networkserver/mac/utils.go#L261-L270

https://github.com/TheThingsNetwork/lorawan-stack/blob/4466080950116b2cc50cecdaef57f9ce0da3a9d3/pkg/networkserver/mac/utils.go#L250-L259

For ABP, this is what happens:

1. If there is no `rx1-delay` in device create, NS assumes RX1 delay of 1 second on L257: https://github.com/TheThingsNetwork/lorawan-stack/blob/4466080950116b2cc50cecdaef57f9ce0da3a9d3/pkg/band/default.go#L24
2. The `desired-rx1-delay` of the Network Server config will then be used to instruct the end device to use a higher RX1 delay. We use 5 seconds in TTN and TTSC.

So, in ABP, the end device is assumed to start with RX1 delay of 1 second and then gets instructed to use 5 seconds. This isn't desirable.

With OTAA this isn't a problem, because the desired settings will be used in the join-accept.

I agree that we shouldn't just increase the default value in the front-end. What we can do is:

1. Declare an `rx1-delay` in default settings of the NS, so we can configure this for TTN and TTSC to be 5 seconds
2. Make this configuration available to the Console, so the user sees what's happening by default. The user can then override this

kschiffer, 2021-07-08:
It would indeed be best if the Console would be aware of the various default settings of the NS, so we can already display them inside the fields. This is also outlined in the wireframes of #3522.

If we decide to do that, please unassign @bafonins and me until the necessary backend changes were made.

johanstokking, 2021-07-08:
@neoaggelos please start with exposing the default NS settings through API per band and PHY version. That's not just exposing `defaults`, but the evaluated result of `DeviceDesiredRx1Delay(nil, phy, defaults)` and the likes.

Please don't close this issue but reassign frontend devs when that's done.

As follow up, @bafonins @kschiffer, you can let the user toggle between desired and default settings for ABP device creation. In the case of the RX1 delay, desired would be 5, default would be 1 second.

neoaggelos, 2021-07-14:
API has been merged, example request/response:

```bash
$ curl http://localhost:1885/api/v3/ns/default_mac_settings/EU_863_870/1.0.2-b
{
  "class_b_timeout": "600s",
  "ping_slot_data_rate_index": 3,
  "ping_slot_frequency": "869525000",
  "beacon_frequency": "0",
  "class_c_timeout": "300s",
  "rx1_delay": 1,
  "rx1_data_rate_offset": 0,
  "rx2_data_rate_index": 0,
  "rx2_frequency": "869525000",
  "max_duty_cycle": "DUTY_CYCLE_1",
  "supports_32_bit_f_cnt": true,
  "use_adr": true,
  "adr_margin": 15,
  "resets_f_cnt": false,
  "status_time_periodicity": "86400s",
  "status_count_periodicity": 200,
  "desired_rx1_delay": 5,
  "desired_rx1_data_rate_offset": 0,
  "desired_rx2_data_rate_index": 0,
  "desired_rx2_frequency": "869525000",
  "desired_max_duty_cycle": "DUTY_CYCLE_1",
  "desired_adr_ack_limit_exponent": "ADR_ACK_LIMIT_64",
  "desired_adr_ack_delay_exponent": "ADR_ACK_DELAY_32",
  "desired_ping_slot_data_rate_index": 3,
  "desired_ping_slot_frequency": "869525000",
  "desired_beacon_frequency": "0",
  "desired_max_eirp": "DEVICE_EIRP_16"
}
```

kschiffer, 2021-07-27:
So in the registration form, we will deal only with fields that **need** to be set during registration. This means that we will only deal with non-`desired_*` fields during registration. So necessary changes are

*   Use fetched band values as default value for fields
*   Whenever there is a desired value that differs from the default, display this value next to the respective field
    *   Also, render a tooltip next to the value that explains about desired values and their implications, cc @johanstokking please help with this
*   Make every field with a default value required
*   In a separate PR: add a way to set desired mac parameters in the end device general settings, using the same mechanics as described above.

johanstokking, 2021-07-27:
Tooltips

* **RX1 delay**: Time in seconds after an uplink transmission until the first receive window (RX1) opens. The second receive window (RX1) opens exactly 1 second after RX1. OTAA devices receive the RX1 delay as part of the join-accept. Changing this value during an OTAA or ABP session makes the Network Server transmit the `RXTimingSetupReq` MAC command.
* **RX1 data rate offset**: Used to lower data rate in the first receive window (RX1). When using 0, there is no offset applied. A value higher than 0 reduces the RX1 data rate. See LoRaWAN Regional Parameters to lookup the downstream data rate in RX1 for each offset value. OTAA devices receive the RX1 data rate offset as part of the join-accept. Changing this value during an OTAA or ABP session makes the Network Server transmit the `RXParamSetupReq` MAC command. 
* **RX2 data rate index**: The second receive window (RX2) uses a fixed frequency and data rate. This value configures the data rate to use in RX2. See LoRaWAN Regional Parameters to find the band's modulation settings for each data rate index. OTAA devices receive the RX1 data rate offset as part of the join-accept. Changing this value during an OTAA or ABP session makes the Network Server transmit the `RXParamSetupReq` MAC command.
* **RX2 data rate frequency**: The second receive window (RX2) uses a fixed frequency and data rate. This value configures the frequency to use in RX2. Changing this value makes the Network Server transmit the `RXParamSetupReq` MAC command.
* **Maximum duty cycle**: The maximum aggregated transmit duty cycle of the end device over all sub-bands. The allowed time-on-air is 1/N where N is the given value. 0 means no duty cycle limit. This value is used for traffic shaping. All end devices must respect regional regulations regardless of this value. Changing this value makes the Network Server transmit the `DutyCycleReq` MAC command. 
* **ADR ACK limit**: This value changes the `ADR_ACK_LIMIT` value defining the ADR back-off algorithm. Changing this value makes the Network Server transmit the `ADRParamSetupReq` MAC command (from LoRaWAN 1.1).
* **ADR ACK delay**: This value changes the `ADR_ACK_DELAY` value defining the ADR back-off algorithm. Changing this value makes the Network Server transmit the `ADRParamSetupReq` MAC command (from LoRaWAN 1.1).
* **Ping slot data rate index**: The class B ping slot uses a fixed frequency and data rate. This value configures the data rate to use in class B ping slots. Changing this value makes the Network Server transmit the `PingSlotChannelReq` MAC command.
* **Ping slot frequency**: The class B ping slot uses a fixed frequency and data rate. This value configures the frequency to use in class B ping slots. Changing this value makes the Network Server transmit the `PingSlotChannelReq` MAC command.
* **Beacon frequency**: The class B beacon is sent on a fixed frequency. This value changes the frequency to use in class B beacons. Use the same setting for all end devices using class B in the network. Changing this value makes the Network Server transmit the `BeaconFreqReq` MAC command.
* **Maximum EIRP**: The maximum allowed Effective Isotropic Radiated Power (EIRP). The end device uses this value as the maximum: it shall never use a higher EIRP, but a lower EIRP is allowed. This value is given in dBm. Changing this value makes the Network Server transmit the `TxParamSetupReq`.

> * In a separate PR: add a way to set desired mac parameters in the end device general settings, using the same mechanics as described above.

OK, please file follow up issue.

johanstokking, 2021-07-28:
ABP current vs desired settings tooltip on registration:

An ABP device is personalized with a session and MAC settings. These MAC settings are considered the **current parameters** and must match exactly the settings entered here. Leave empty to use LoRaWAN defaults. The Network Server uses **desired parameters** to change the MAC state with LoRaWAN MAC commands to the desired state. The desired values are shown.

---

ABP required fields on registration:

- Class B enabled
	- Class B timeout
	- Ping slot data rate index
	- Ping slot frequency
	- Beacon frequency
- Class C enabled
	- Class C timeout
- RX1 delay
- RX1 data rate offset
- RX2 data rate index
- RX2 frequency
- Resets FCnt

---

OTAA required fields on registration:

- Class B enabled
- Class C enabled

---

Settings and desired parameters for both ABP and OTAA:

- Class B enabled
	- Class B timeout
	- Ping slot data rate index
	- Ping slot frequency
	- Beacon frequency
- Class C enabled
	- Class C timeout
- RX1 delay
- RX1 data rate offset
- RX2 data rate index
- RX2 frequency

- Maximum duty cycle
- ADR enabled
	- ADR margin
	- ADR ACK limit (LoRaWAN 1.1)
	- ADR ACK delay (LoRaWAN 1.1)
- Status time periodicity
- Status count periodicity

Only for ABP:
- Resets FCnt

bafonins, 2021-08-22:
@neoaggelos based on your comment https://github.com/TheThingsNetwork/lorawan-stack/issues/4290#issuecomment-879693986 NS returns `0` for `beacon_frequency` entry as the default value which is invalid for NS. When submitting this to NS the following error is returned:
```
{
  "code": 3,
  "message": "error:pkg/errors:validation (invalid `end_device`: embedded message failed validation)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/errors",
      "name": "validation",
      "message_format": "invalid `{field}`: {reason}",
      "attributes": {
        "field": "end_device",
        "name": "SetEndDeviceRequestValidationError",
        "reason": "embedded message failed validation"
      },
      "correlation_id": "b77efa0a4e2b4acebb93c8e4ffb23350",
      "cause": {
        "namespace": "pkg/errors",
        "name": "validation",
        "message_format": "invalid `{field}`: {reason}",
        "attributes": {
          "field": "mac_settings",
          "name": "EndDeviceValidationError",
          "reason": "embedded message failed validation"
        },
        "correlation_id": "965ecddea4ec42e4996bc875782831d5",
        "cause": {
          "namespace": "pkg/errors",
          "name": "validation",
          "message_format": "invalid `{field}`: {reason}",
          "attributes": {
            "field": "beacon_frequency",
            "name": "MACSettingsValidationError",
            "reason": "embedded message failed validation"
          },
          "correlation_id": "f74c901da0ae48bbaaefcd3732a93d53",
          "cause": {
            "namespace": "pkg/errors",
            "name": "validation",
            "message_format": "invalid `{field}`: {reason}",
            "attributes": {
              "field": "value",
              "name": "FrequencyValueValidationError",
              "reason": "value must be greater than or equal to 100000"
            },
            "correlation_id": "1fbdacb26bdf4033a12e2f27cd42671f",
            "code": 3
          },
          "code": 3
        },
        "code": 3
      },
      "code": 3
    }
  ]
}
```

Request payload:
```
{
  "end_device": {
    "frequency_plan_id": "EU_863_870",
    "lorawan_phy_version": "PHY_V1_0",
    "multicast": false,
    "supports_join": false,
    "lorawan_version": "MAC_V1_0",
    "ids": {
      "device_id": "eui-1237812371238978",
      "dev_eui": "1237812371238978"
    },
    "session": {
      "keys": {
        "f_nwk_s_int_key": {
          "key": "5D70FE1C76C1147C165C7669629FDFDE"
        }
      },
      "dev_addr": "01ACC107"
    },
    "supports_class_c": false,
    "supports_class_b": true,
    "mac_settings": {
      "class_b_timeout": "600s",
      "ping_slot_periodicity": "PING_EVERY_1S",
      "ping_slot_data_rate_index": 3,
      "rx2_data_rate_index": 0,
      "ping_slot_frequency": 869525000,
      "beacon_frequency": 0,
      "rx2_frequency": 869525000,
      "rx1_delay": 1,
      "rx1_data_rate_offset": 0,
      "resets_f_cnt": false
    }
  },
  "field_mask": {
    "paths": [
      "frequency_plan_id",
      "lorawan_phy_version",
      "multicast",
      "supports_join",
      "lorawan_version",
      "ids.device_id",
      "ids.dev_eui",
      "session.keys.f_nwk_s_int_key.key",
      "session.dev_addr",
      "supports_class_c",
      "supports_class_b",
      "mac_settings.class_b_timeout",
      "mac_settings.ping_slot_periodicity",
      "mac_settings.ping_slot_data_rate_index",
      "mac_settings.rx2_data_rate_index",
      "mac_settings.ping_slot_frequency",
      "mac_settings.beacon_frequency",
      "mac_settings.rx2_frequency",
      "mac_settings.rx1_delay",
      "mac_settings.rx1_data_rate_offset",
      "mac_settings.resets_f_cnt"
    ]
  }
}
```

Is this expected?

neoaggelos, 2021-08-24:
This is because we always set the `BeaconFrequency` (which is a `ttnpb.FrequencyValue`) even if `mac.DeviceDefaultBeaconFrequency()` returns 0.

An easy workaround for `BeaconFrequency` (and `DesiredBeaconFrequency`) is https://github.com/TheThingsNetwork/lorawan-stack/commit/293ee1a744ed5944232df75eb8c711a90f531d7b. cc @adriansmares

closed 2021-11-24 by bafonins