TheThingsNetwork/lorawan-stack #4294: Error while starting the things stack "Cannot connect to idenity server"

justinbyerss, 2021-06-22:
#### Summary

Error connecting to identity server database
...

#### Steps to Reproduce

Need help to rectify

[ttn-lw-stack-docker.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6691269/ttn-lw-stack-docker.txt)
[image]

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?

...

#### Environment

...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

neoaggelos, 2021-06-22:
Hi @justinbyerss

This looks like an issue with the database container. Can you see if the database is running?

What's the output of the following commands?

```bash
$ sudo docker-compose ps
$ sudo docker-compose logs cockroach
```

It might help if you also run

```bash
$ sudo docker-compose up -d redis cockroach
```

So that Redis and CockroachDB have time to properly start and become ready to accept connections.

justinbyerss, 2021-06-22:
> Hi @justinbyerss
> 
> This looks like an issue with the database container. Can you see if the database is running?
> 
> What's the output of the following commands?
> 
> ```shell
> $ sudo docker-compose ps
> $ sudo docker-compose logs cockroach
> ```
> 
> It might help if you also run
> 
> ```shell
> $ sudo docker-compose up -d redis cockroach
> ```
> 
> before running any other commands, so that Redis and CockroachDB have time to properly start and become ready to accept connections.

Thanks for your assistance.  I did a clean install on my PC in hope to resolve the issue.  I have managed to get past that issue but now I cannot get past this. 
[image]

neoaggelos, 2021-06-22:
How are you trying to connect to the Console (`http` or `https`)? Also, which Docker tag are you using for the `thethingsnetwork/lorawan-stack` image?

justinbyerss, 2021-06-22:
> How are you trying to connect to the Console (`http` or `https`)? Also, which Docker tag are you using for the `thethingsnetwork/lorawan-stack` image?

I am using the latest tag for all three containers.  I just tried using https and it worked.  It didn't seem to like Http.

neoaggelos, 2021-06-24:
This should be fixed with https://github.com/TheThingsNetwork/lorawan-stack/pull/4303.

Closing the issue, please re-open if the issue persists with the new version.

closed 2021-06-24 by neoaggelos