TheThingsNetwork/lorawan-stack #4306: Add winget (Windows package manager) support for installing lorawan-stack-cli

lnlp, 2021-06-23:
#### Summary
Add support for Microsoft's official Windows package manager **winget** for installing lorawan-stack-cli.

#### Why do we need this?

So that Windows users can simply install the TTN CLI tools with: `winget install ttn-lw-cli` 

Similar to how macOS and Linux users can currently install it via their package managers.

Packages for macOS and Linux package managers are already supported so the importance of package managers does not need further explanation (neither does the importance of the Windows operating system).

Until recently there was no official package manager for Windows but now there is an official Windows (10) package manager from Microsoft called: **winget**

`winget` currently needs to be manually installed but will be added standard to Windows in a near future Windows update. 
To install `winget` paste below URL (exactly as shown) in a web browser on Windows 10 (this will update the Windows App Installer with a version that includes the `winget` CLI tool).

```text
ms-appinstaller:?source=https://aka.ms/getwinget
```

#### What is already there? What do you see now?

Unlike for macOS and Linux the `winget` package manager for Windows is currently not supported.

#### What is missing? What do you want to see?

Support for the `winget` package manager to install the `lorawan-stack-cli` (`ttn-lw-cli`).

#### Environment

Windows 10.

#### How do you propose to implement this?

Information about the Windows Package Manager Community Repository and how to create a package can be found here:
https://github.com/microsoft/winget-pkgs

#### How do you propose to test this?

Needs no further explanation.

#### Can you do this yourself and submit a Pull Request?

No.

johanstokking, 2021-06-28:
Thanks @lnlp !

neoaggelos, 2021-07-07:
Currently, winget requires an installer, which we do not have for Windows builds. There is an ongoing discussion to support installing from `.zip` natively with winget in [this issue](https://github.com/microsoft/winget-cli/issues/140), along with recent activity.

Alternatively:

- there was also some discussion in this [goreleaser issue](https://github.com/goreleaser/goreleaser/issues/1295), but that one does not seem very active.
- we could create a very simple installer using something like NSIS (but I am not sure how well that can integrate with our github actions workflows atm)

In general, I do like the idea of having `winget install ttn-lw-cli`, but it is currently not as straightforward as declaring a YAML manifest. My suggestion would be to wait until [Winget adds support for installing from .zip files](https://github.com/microsoft/winget-cli/issues/140).

If this takes a long time, and people from the community want this sooner, then we can prioritize this and see what the best way would be to create a simple installer for ttn-lw-cli.

**In short, this issue is currently blocked. If you are part of the community and would like `ttn-lw-cli` to get winget support sooner, please upvote this issue.**