TheThingsNetwork/lorawan-stack #4326: Location setting doesn't allow longitude of selected position

aparcar, 2021-06-28:
#### Summary

I setup a node on Oahu, Hawaii and tried to add it's location via the web interface. Clicking at the destination results in a Longitude of ~200 something, while it's real location is on Longitude 157.x. The web interface doesn't support the input (must be within +/-180).

#### Steps to Reproduce

Try to use the map to set the location somewhere on the Hawaiian Islands.

#### What do you see now?

[image]

Setting the Longitude manually to the correct value results in a destination within the ocean (wrong):

[image]

#### What do you want to see instead?

The correct Longitude.

#### Environment

Latest Firefox.

#### How do you propose to implement this?

Fix the map data provider

#### How do you propose to test this?

-

#### Can you do this yourself and submit a Pull Request?

Not really, I'm not familiar with OSM/Leaflet.

nejraselimovic, 2021-06-29:
@bafonins please check if there is a quick fix for this and if it can by any case be done by the v3.13.3 deadline

bafonins, 2021-06-29:
Hey @aparcar! This issue will be fixed in the next release.

aparcar, 2021-06-29:
Nice rtt, thanks