TheThingsNetwork/lorawan-stack #4334: Gateway positions not updating from status messages

virtualguy, 2021-07-01:
#### Summary

The gateway location information doesnt update even though the gps positions are  available in the gateway status and the uplinks topic under antenna location

#### Steps to Reproduce

1. Create gateway
2. Connect gateway and send gateway status
3. Gateway position doesnt update in the console or from ttn-lw-cli

#### What do you see now?

```
{
  "name": "gs.status.receive",
  "time": "2021-07-01T03:59:18.502649297Z",
  "identifiers": [
    {
      "gateway_ids": {
        "gateway_id": "eui-647fdafffe00a4eb",
        "eui": "647FDAFFFE00A4EB"
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.GatewayStatus",
    "time": "2021-07-01T03:59:18Z",
    "boot_time": "0001-01-01T00:00:00Z",
    "versions": {
      "ttn-lw-gateway-server": "3.13.1"
    },
    "antenna_locations": [
      {
        "latitude": -36.85179,
        "longitude": 174.77522,
        "altitude": 27
      }
    ],
    "ip": [
      "43.243.56.33"
    ],
    "metrics": {
      "ackr": 100,
      "rxfw": 0,
      "rxin": 1,
      "rxok": 0,
      "temp": 47,
      "txin": 0,
      "txok": 0
    }
  },
  "correlation_ids": [
    "gs:conn:01F9G2TYKJT082EP1KW061Z437",
    "gs:status:01F9G3H4Z66JX84H24BC9BB3PK"
  ],
  "origin": "884358cb38cf",
  "visibility": {
    "rights": [
      "RIGHT_GATEWAY_STATUS_READ"
    ]
  },
  "unique_id": "01F9G3H4Z6DBXFV8BWF0XRRM0H"
}
```

[image]

#### What do you want to see instead?
Latitude, longitude and altitude displayed in the console and available from

#### Environment

ttn-lw-stack 3.13.1

#### How do you propose to implement this?

#### How do you propose to test this?

I can test branches or docker builds

#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2021-07-01:
@virtualguy thank you for filing the issue! What type of gateway are you using? If it is a UDP gateway, check [this](https://github.com/TheThingsNetwork/lorawan-stack/pull/3326) and [this](https://github.com/TheThingsNetwork/lorawan-stack/issues/3138)

virtualguy, 2021-07-01:
Yes indeed this is for a UDP packet forwarder. Are there any work arounds or is this just not supported

nejraselimovic, 2021-07-01:
> Yes indeed this is for a UDP packet forwarder. Are there any work arounds or is this just not supported

@neoaggelos could better answer that question

nejraselimovic, 2021-07-02:
I'm closing this issue since this behavior is intended. @neoaggelos please let us know if there is a workaround

closed 2021-07-02 by nejraselimovic

kurtmc, 2021-07-04:
So the intended behaviour is for the API to return `null` for `antenna_locations` when "Update from status messages" is checked?