TheThingsNetwork/lorawan-stack #4335: Gateway console field 'LoRa Basics Station LNS Authentication Key' is empty after recovering from "An unknown error occurred"

bertrik, 2021-07-01:
#### Summary

In the gateway console, the field 'LoRa Basics Station LNS Authentication Key' is empty after recovering from "An unknown error occurred".

#### Steps to Reproduce

1. Open the TTN v3 gateway console and select a gateway (in my case a freshly migrated TTIG)
2. Wait a bit (how long is not very clear yet, about 2 minutes seems to trigger it)
3. Click on "General Settings" -> Message "An unknown error occurred" is shown
4. Refresh in the browser -> Gateway general settings are shown again, however field "LoRa Basics Station LNS Authentication Key" is now empty.

#### What do you see now?
[image]

#### What do you want to see instead?
I'd like to update other properties of the gateway on the console.
If the LNS key is not filled somehow, I'm afraid of accidentally removing it when saving to update other properties.

#### Environment
OS: windows 10
Device: TTIG
Browser: Chrome Versie 91.0.4472.124 (Officiële build) (64-bits)

kschiffer, 2021-07-06:
@KrishnaIyer is this fixed by #4355 ?

KrishnaIyer, 2021-07-06:
I think it could more be related to https://github.com/TheThingsNetwork/lorawan-stack/pull/4359. Let's get that and https://github.com/TheThingsNetwork/lorawan-stack/pull/4355 deployed and revisit this issue if it's still relevant.

KrishnaIyer, 2021-07-30:
I think I found the issue. This check returns false and the `lbs_lns_secret` is not added to the list of request fields to the backend;
https://github.com/TheThingsNetwork/lorawan-stack/blob/6d72ec73797051985a65dac80bb8e55590e6d313/pkg/webui/console/views/gateway/index.js#L76

@bafonins do you know why this is? My JS isn't good enough to trace this.

closed 2021-08-04 by bafonins