TheThingsNetwork/lorawan-stack #4345: Update Snapcraft login

johanstokking, 2021-07-02:
#### Summary

Update Snapcraft login

#### Why do we need this?

The current one expires end of January 2022.

#### What is already there? What do you see now?

Current login file:

```
snaps:       ['ttn-lw-stack']
channels:    ['stable']
permissions: ['package_access', 'package_manage', 'package_push', 'package_register', 'package_release', 'package_update']
expires:     2022-02-01T00:00:00.000008
```

#### What is missing? What do you want to see?

Bump another year to expire end of January 2023 or another convenience moment.

#### How do you propose to implement this?

```bash
$ snapcraft export-login --snaps ttn-lw-stack --channels stable --expires "DATE" snapcraft.login
```

#### How do you propose to test this?

Test login with:

```bash
$ snapcraft login --with snapcraft.login
```

#### Can you do this yourself and submit a Pull Request?

Yes

htdvisser, 2021-11-12:
Updated.

```
snaps:       ['ttn-lw-stack']
channels:    ['edge', 'beta', 'candidate', 'stable']
permissions: ['package_access', 'package_manage', 'package_metrics', 'package_push', 'package_register', 'package_release', 'package_update']
expires:     2022-11-12T12:07:23.226769
```

Apparently max expiry is a year, so it didn't accept January 2023.

KrishnaIyer, 2024-06-26:
@johanstokking do we know what the current expiry is? cc: @happyRip

johanstokking, 2024-06-26:
The last update of the repository Actions secret `SNAPCRAFT_LOGIN` is 4 months ago.

KrishnaIyer, 2024-07-01:
Ok then we need to check back here in 8 months. I'll add this to the backlog milestone so we revisit it every 3 months.