TheThingsNetwork/lorawan-stack #4347: Add band identifier and version to TxRequest

adriansmares, 2021-07-02:
#### Summary

References https://github.com/TheThingsNetwork/lorawan-stack/issues/1394

We should add the band identifiers and `PHYVersion` to `TxRequest`

#### Why do we need this?

As signaled in https://github.com/TheThingsNetwork/lorawan-stack/issues/4196 , data rate indices may map to different SF/BW across different regional parameters version. As such, we need the band identifier and version in order to allow the Gateway Server to translate the indices to the right values.

#### What is already there? What do you see now?

`rx1_data_rate_index` and `rx2_data_rate_index`, but no `PHYVersion` and band id.

#### What is missing? What do you want to see?

```proto
PHYVersion band_version = 11;
string frequency_plan_id = 12 [(validate.rules).string.max_len = 64];
```

The Network Server shall provide the band version and frequency plan ID, while the Gateway Server shall use them to derive the SF/BW.

#### Environment

`v3.14`

#### How do you propose to implement this?

- Add the proto definitions
- The NS should fill in the version and ID while generating the `TxRequest`
https://github.com/TheThingsNetwork/lorawan-stack/blob/2e5e9dfab39b7adb909f09a7a7e366c57dee6395/pkg/networkserver/downlink.go#L1236
https://github.com/TheThingsNetwork/lorawan-stack/blob/2e5e9dfab39b7adb909f09a7a7e366c57dee6395/pkg/networkserver/downlink.go#L1409
- The PBA should fill these two (issue inc)
https://github.com/TheThingsNetwork/lorawan-stack/blob/2e5e9dfab39b7adb909f09a7a7e366c57dee6395/pkg/packetbrokeragent/translation.go#L625
- The GS should take them into account while translating the data rate index
https://github.com/TheThingsNetwork/lorawan-stack/blob/2e5e9dfab39b7adb909f09a7a7e366c57dee6395/pkg/gatewayserver/io/io.go#L440

#### How do you propose to test this?

Test the SF/BW of a downlink sent via AU915 1.0.0 downlink.

#### Can you do this yourself and submit a Pull Request?

I can do the API and NS part, cc @KrishnaIyer for GS and @johanstokking for PBA.

johanstokking, 2021-07-02:
Cool, @KrishnaIyer please start with API soon, don't close this issue but re-assign once API is there:

- @KrishnaIyer to implement in GS
- @adriansmares to implement in NS
- @johanstokking to implement in PBA (needs https://github.com/packetbroker/api/issues/31)

KrishnaIyer, 2021-07-05:
> @KrishnaIyer please start with API soon

Isn't API already here: https://github.com/TheThingsNetwork/lorawan-stack/pull/4349?

KrishnaIyer, 2021-07-13:
This is blocked on https://github.com/TheThingsNetwork/lorawan-stack/issues/4346.

With #4349, we have the `lorawan_phy_version` but the GS can't really use it until https://github.com/TheThingsNetwork/lorawan-stack/issues/4346.

johanstokking, 2021-07-13:
The band can already be versioned. #4346 is about maintaining the frequency plans in a versioned way, but that won't really change the way GS uses it.

KrishnaIyer, 2021-07-16:
GS done via #4406. I guess only PBA remains. @johanstokking please close the issue if that's already done.

johanstokking, 2021-07-16:
PBA still needs to happen

closed 2021-07-20 by johanstokking