TheThingsNetwork/lorawan-stack #4360: Introduce NS-AS communication fast path

adriansmares, 2021-07-06:
#### Summary

The NS should skip queueing messages to the AS if the associated peer is available.

Currently, uplinks that should be submitted to the AS are always queued in a Redis stream, which is then asynchronously drained. This introduces quite some latency even though the peer is available and the stream could have been skipped.

The 'quite some latency' mentioned above is based on the metrics from https://github.com/TheThingsNetwork/lorawan-stack/pull/4258 / https://github.com/TheThingsNetwork/lorawan-stack/pull/4258 . The latency is heavily correlated to the number of sources (NS) and sinks available (AS instances).

#### Why do we need this?

To lower the NS-AS latency, in order to allow integrations to respond to uplink messages within `RX1`.

#### What is already there? What do you see now?

The uplinks are enqueued via `ns.enqueueApplicationUplinks`. The queue is then drained via `ns.processApplicationUplinkTask`.

#### What is missing? What do you want to see?

I would like to see a 'fast-path' in `ns.enqueueApplicationUplinks` that attempts to contact the peer directly and submit the uplinks, instead of always enqueuing them. If the direct drain fails, only then the uplinks should be enqueued.

#### Environment

`v3.14`.

#### How do you propose to implement this?

`ns.processApplicationUplinkTask` already contains most of the logic used for drainage. We can extract it from there and use it on both paths (fast path / slow path).

#### How do you propose to test this?

Unit tests should already cover this, and the latency gains can be tested via the metrics from the PRs above.

#### Can you do this yourself and submit a Pull Request?

Yes. Pinging @johanstokking for an opinion on this - if all is clear please remove the `discuss` label.

adriansmares, 2021-07-06:
References https://github.com/TheThingsNetwork/lorawan-stack/issues/4361 - currently there is only one drain per Network Server instance that submits uplinks to an Application Server, which also hurts latency.

johanstokking, 2021-07-07:
Sounds good!

PR #4485 (2021-08-03): Skip the application uplink queue if the peer is available

#### Summary

Closes #4360 

#### Changes

- The Network Server application uplink queue is now skipped if the Network Server successfully retrieves the Application Server peer and submits the uplinks.
  - The submission is done in a separate task. The concurrency is limited to 1024 tasks. When all tasks have been used, the NS automatically enqueues the uplinks.

#### Testing

Unit testing and local testing.

##### Regressions

We'll have more uplinks in memory after this change, but that's still faster than the queue path.

#### Chec [... cut]
closed 2021-08-03 by adriansmares