TheThingsNetwork/lorawan-stack #4361: Spawn multiple tasks in the Network Server

adriansmares, 2021-07-06:
#### Summary

We currently spawn only one task per NS instance in order to schedule downlinks and to drain uplinks. 

#### Why do we need this?

These are more or less IO-bound operations, so we should spawn more in order to improve latency.
This is also safe, as the tasks are designed to be run in parallel (from any NS Instance).

#### What is already there? What do you see now?

https://github.com/TheThingsNetwork/lorawan-stack/blob/1a69b973744552c8df0fbaaf33f7846aae430e00/pkg/networkserver/networkserver.go#L249-L260

#### What is missing? What do you want to see?

Multiple tasks for each type (downlink processing / uplink drainage).

#### Environment

`v3.14`.

#### How do you propose to implement this?

Introduce a setting (for me, only one suffices) that allows us to configure how many tasks to have running at once, and wrap the for loop above in order to start `n` tasks of each time.

#### Can you do this yourself and submit a Pull Request?

Leaving this one up to @neoaggelos . cc @johanstokking if you see any issue with this.

adriansmares, 2021-07-06:
This is not `size/small` on a second look into how the queues work.

Currently the consumer ID is global per Network Server instance, meaning that calling 2 `Pop`s in parallel will return the same records twice.

In order for this to work cleanly, we need to force the caller to provide the consumer ID. This requires the following changes:

- Queues (`ttnredis.TaskQueue`, `nsredis.ApplicationUplinkQueue`, `nsredis.DownlinkTaskQueue`) should no longer require the consumer ID to be provided on initialization.
- `ttnredis.TaskQueue` should maintain an internal `sync.Map` between the seen consumer IDs and a `struct{}`. This acts as a set of 'used' consumer IDs.`Close` then can range over this set and delete the consumers.
- All of the implementations should be forced to provide their consumer ID while attempting to `Pop`. This consumer ID is then to be used instead of the internal one.

Choosing the consumer IDs on the task side should be related to how we do now - a combination of hostname, PID, and now task ID (not the whole name, just the task number). `hostname:PID:number` should suffice (no need to use the `ttnredis` function).

htdvisser, 2021-08-04:
Since @neoaggelos isn't going to finish this before the v3.14.1 deadline, I'll move it to v3.14.2.

PR #4525 (2021-08-24): Multiple Network Server task consumers 

#### Summary

Closes #4361.

Introduces new configuration, but defaults are fully backwards compatible.

#### Changes

Pretty much the issue description.

- Explicitly specify consumer ID in calls to `TaskQueue.Pop()`, and update `nsredis.ApplicationUplinkQueue`, `nsredis.DownlinkTaskQueue` accordingly.
- Add configuration for spawning multiple task queues in the same Network Server instance.
- Update unit tests to test with multiple concurrent consumers. Some minor changes to the test specification, see inline comments.

#### Testing

Unit tests, also  [... cut]
closed 2021-08-24 by adriansmares