TheThingsNetwork/lorawan-stack #4375: Can’t exchange authorization code for an OAuth access token

romach, 2021-07-07:
#### Summary

If OAuth client has `GRANT_AUTHORIZATION_CODE` and `GRANT_REFRESH_TOKEN` grants, it is not possible to exchange authorization code for an OAuth access token

#### Steps to Reproduce

1. Create OAuth client with `GRANT_AUTHORIZATION_CODE` and `GRANT_REFRESH_TOKEN` grants
    ```
    ttn-lw-cli clients create tektelic-refresh-token --description "Description"  --organization-id tektelic-dev --redirect-uris <redirect_url> --rights RIGHT_APPLICATION_ALL,RIGHT_GATEWAY_ALL --grants GRANT_AUTHORIZATION_CODE,GRANT_REFRESH_TOKEN
    ```
2. Get authorization code opening page
    ```
    https://eu1.cloud.thethings.network/oauth/authorize?client_id=tektelic-refresh-token&redirect_uri=<redirect_uri>&state=roman_cherepanov&response_type=code
    ```
3. Try to exchange authorization code for an OAuth access token using `POST` request to `https://eu1.cloud.thethings.network/oauth/token`, use Basic Auth with the client ID as username and the client secret as password. Use payload `{ "code": "<AUTHORIZATION-CODE>", "grant_type": "authorization_code"}`

#### What do you see now?

```
{
    "code": 7,
    "message": "error:pkg/oauth:invalid_grant (invalid, expired or revoked authorization code)",
    "details": [
        {
            "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
            "namespace": "pkg/oauth",
            "name": "invalid_grant",
            "message_format": "invalid, expired or revoked authorization code",
            "code": 7
        }
    ]
}
```

#### What do you want to see instead?

```
{
    "access_token": "<...>",
    "expires_in": 3600,
    "token_type": "bearer",
    "refresh_token": "<...>",
}
```

#### Environment

```
ttn-lw-cli version
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.13.2
Build date:          2021-06-17T12:36:42Z
Git commit:          4a2e92c4c
Go version:          go1.16.5
OS/Arch:             darwin/amd64
```

#### Additional information

I OAuth client contains only `GRANT_AUTHORIZATION_CODE` grant, I can exchange authorization code for an OAuth access token.

nejraselimovic, 2021-07-07:
@neoaggelos I'm not sure if you're the one who can help with this, please reassign if not

neoaggelos, 2021-07-07:
Hi @romach, thank you for filing the issue.

OAuth clients on need to be approved by an admin first before they can be used. Can you share the output of the following command? (the state in particular)

```bash
$ ttn-lw-cli clients get tektelic-refresh-token --state --rights --grants
```

romach, 2021-07-07:
@neoaggelos 

Here is a response of the command

```
ttn-lw-cli clients get tektelic-refresh-token --state --rights --grants
```

```
{
  "ids": {
    "client_id": "tektelic-refresh-token"
  },
  "created_at": "2021-07-05T05:55:39.372Z",
  "updated_at": "2021-07-06T13:15:33.490Z",
  "state": "STATE_APPROVED",
  "grants": [
    "GRANT_AUTHORIZATION_CODE",
    "GRANT_REFRESH_TOKEN"
  ],
  "rights": [
    "RIGHT_APPLICATION_ALL",
    "RIGHT_GATEWAY_ALL"
  ]
}
```

neoaggelos, 2021-07-07:
Great, thank you, then indeed it could be a bug.

Assigning @htdvisser who is well familiar with the internals of the oauth server.

htdvisser, 2021-07-08:
Looking at the server logs I see the following related requests:

```

2021-07-06T14:25:40,"method=GET host="eu1.cloud.thethings.network" path="/oauth/authorize?client_id=tektelic-refresh-token&redirect_uri=https%3A%2F%2Fapi-dev.onboarding.tektelic-dev.com%2Fapi%2Fttn%2Foauth%2Fcallback&response_type=code&state=roman_cherepanov" request_id="e108e548-df50-43aa-a0e8-57f0b8306780"
2021-07-06T14:26:10,"method=POST host="eu1.cloud.thethings.network" path="/oauth/authorize?client_id=tektelic-refresh-token&redirect_uri=https%3A%2F%2Fapi-dev.onboarding.tektelic-dev.com%2Fapi%2Fttn%2Foauth%2Fcallback&response_type=code&state=roman_cherepanov" request_id="944eb879-5ac7-4bf3-baa2-c1c0a5cc5ba9"
2021-07-06T14:38:30,"method=POST host="eu1.cloud.thethings.network" path="/oauth/token" request_id="bbb43b6a-9304-4d74-b434-b0906055b188"
2021-07-06T14:39:10,"method=POST host="eu1.cloud.thethings.network" path="/oauth/token" request_id="ab472d9b-f1e0-4499-bafd-d5bc5dabe9ed"
2021-07-06T14:39:50,"method=POST host="eu1.cloud.thethings.network" path="/oauth/token" request_id="e5f92c95-b930-47cd-a315-53b926d78a8d"
2021-07-06T14:40:10,"method=POST host="eu1.cloud.thethings.network" path="/oauth/token" request_id="3e1db59a-118f-4910-8c5b-8eace2b87c69"
2021-07-06T14:40:20,"method=POST host="eu1.cloud.thethings.network" path="/oauth/token" request_id="3f8d75f3-e13d-4b24-b758-a7269f32fb62"
2021-07-06T14:40:30,"method=POST host="eu1.cloud.thethings.network" path="/oauth/token" request_id="d0223c03-13b3-4f83-b238-f48b1089956f"
2021-07-06T14:41:50,"method=POST host="eu1.cloud.thethings.network" path="/oauth/token" request_id="7b943cb8-bce5-420c-95e6-7d65107652f3"
2021-07-06T14:42:00,"method=POST host="eu1.cloud.thethings.network" path="/oauth/token" request_id="acc921e6-2af5-48a5-a262-1914473b80ac"
2021-07-06T14:48:20,"method=POST host="eu1.cloud.thethings.network" path="/oauth/token" request_id="96833bf3-d72c-431d-8bfb-c3ed11a910d3"
```

From this we can see that the authorization code was issued at `2021-07-06T14:26:10`. Authorization codes are valid for 5 minutes, so when a token exchange is attempted at `2021-07-06T14:38:30`, the code already expired.

Therefore this is not a bug, and the returned error message is correct.

closed 2021-07-08 by htdvisser

romach, 2021-07-08:
@htdvisser how do you think, if it will be useful to add information about the Authorization code expiration time to the documentation https://www.thethingsindustries.com/docs/reference/api/authentication/#oauth-access-tokens ?

htdvisser, 2021-07-08:
Considering that it would have been useful to you, I'm sure it would be useful to others. I'll add it.