TheThingsNetwork/lorawan-stack #4378: Generate markdown docs and JSON tree

benolayinka, 2021-07-07:
#### Summary

Generate markdown docs for cobra commands, and a json tree structure, like https://github.com/packetbroker/pb/pull/11

It looks like GenMDDoc already exists, but it probably needs a prepender and link handler to add front matter and turn links in to hugo refs
old ->  https://github.com/TheThingsNetwork/lorawan-stack/blob/2e5e9dfab39b7adb909f09a7a7e366c57dee6395/cmd/internal/commands/doc.go#L64
pb ref -> https://github.com/packetbroker/pb/blob/baafadf27bad8ec1ec779c53dafaacbb516fa037/cmd/internal/gen/doc.go#L21

#### Why do we need this?

To improve CLI docs and tackle https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/271

#### What is already there? What do you see now?

GenMDDoc

#### What is missing? What do you want to see?

GenMDDoc with proper front matter and link handling, and a json tree structure so we can build a map like -> https://packetbroker.github.io/docs/commands/

#### How do you propose to implement this?

@neoaggelos can you take this?

#### Can you do this yourself and submit a Pull Request?

No

benolayinka, 2021-08-05:
Moved to https://github.com/TheThingsNetwork/lorawan-devices/issues/276

benolayinka, 2021-08-30:
@johanstokking @htdvisser could one of you do this without too much trouble? it will also unblock https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/271#issuecomment-872087857

htdvisser, 2021-09-06:
I don't understand what exactly you need. We already generate structured data for all CLI commands and you can already render them with a shortcode.

```
{{< cli/command "applications list" >}}
```

benolayinka, 2021-09-06:
> I don't understand what exactly you need. We already generate structured data for all CLI commands and you can already render them with a shortcode.
> 
> ```
> {{< cli/command "applications list" >}}
> ```

Because I prefer to just use the cobra markdown docs, to create the same cli doc structure we have for packetbroker -> https://packetbroker.net/commands/

But to generate proper markdown docs, cobra needs a prepender (for front matter) and a link substituter (for related cmds), which @johanstokking added here -> https://github.com/packetbroker/pb/blob/baafadf27bad8ec1ec779c53dafaacbb516fa037/cmd/internal/gen/doc.go#L21

And to create the doc tree, @johanstokking exports the tree for me here ->
https://github.com/packetbroker/pb/blob/baafadf27bad8ec1ec779c53dafaacbb516fa037/cmd/internal/gen/tree.go

benolayinka, 2021-09-07:
I've added this feature here

@htdvisser can you make a guess why `GenMDDoc` generates all docs except for the `ttn-lw-cli end-devices templates` command? It's not specific to my branch, this file is also missing when I run `go run ./cmd/ttn-lw-cli gen-md-doc` from v3.14.

closed 2021-10-18 by benolayinka