TheThingsNetwork/lorawan-stack #4379: Rename Last seen to Last activity in the console

KrishnaIyer, 2021-07-08:
#### Summary

Rename Last seen to Last activity in the console 

#### Why do we need this?

The latter is less ambiguous and avoids questions like what exactly is meany by "seen".

#### What is already there? What do you see now?

`Last seen X minutes ago` for gateways and devices. 

#### What is missing? What do you want to see?

`Last activity X minutes ago` for gateways and devices. 

#### Environment

TTS v3.13.3

#### How do you propose to implement this?

I think a description search and replace should do the trick.

#### How do you propose to test this?

Local stack

#### Can you do this yourself and submit a Pull Request?

Yup

KrishnaIyer, 2021-07-30:
Ok this isn't as simple as renaming the message. There are many `LastSeen` variables in the JS code and this might be confusing if I just replace the display text. I'm handing this back to the console guys.

This is not high prio but could be a nice task for a newbie.

kschiffer, 2021-10-05:
While working on this I clean up the UX a bit more by:

1.  Adding tooltips and documentation for the last activity values
2.  Improve resolution of the last seen value (currently not using starting timestamp of sessions)
3.  Remove layout jumps due to changing width of the component
4.  Make pulse animation based on messages coming in

closed 2021-10-15 by kschiffer