TheThingsNetwork/lorawan-stack #4388: List effective routing policies of non-listed networks

johanstokking, 2021-07-10:
#### Summary

List effective routing policies of non-listed networks

#### Why do we need this?

To see all routing policies that apply

#### What is already there? What do you see now?

Currently, the Console fetches first all the (listed) networks, for instance:

`https://tti.eu1.cloud.thethings.industries/api/v3/pba/networks?page=1&limit=20&name_contains=&tenant_id_contains=&with_routing_policy=false`

Then, the Console fetches all the effective routing policies:

`https://tti.eu1.cloud.thethings.industries/api/v3/pba/forwarders/policies?page=1&limit=1000`

The Console then matches the policies of the second response with the networks returned by the first response. This is good and by initial design.

However, there are more effective routing policies in the second response that are not being shown, because those networks are not listed and are therefore not included in the first response.

#### What is missing? What do you want to see?

I would like to see all effective forwarder routing policies.

#### How do you propose to implement this?

I think we should do the following three tabs:

1. **All policies**. This tab should show the result of passing `with_routing_policy=true` to the `/api/v3/pba/networks` call. This will show all networks with an effective routing policy
2. **Non-default policies**. This is the same as what is already there
3. **All listed home networks**. This should show the contents of the current **All** tab

#### How do you propose to test this?

I can configure a few things on staging if that is helpful

#### Can you do this yourself and submit a Pull Request?

Can test

kschiffer, 2021-07-30:
For such unlisted networks, it would not be possible to list the network name (if exists) and dev address blocks, since it is not part of the policy responses. Is there a way we can still get those, or can we include them in the policy responses? Or alternatively, couldn't the network response list unlisted networks as well, in case they have a policy set up towards the own network? I feel like otherwise we're patchworking things a bit.

johanstokking, 2021-08-02:
> Or alternatively, couldn't the network response list unlisted networks as well, in case they have a policy set up towards the own network?

I'll look into this and get back.

johanstokking, 2021-08-04:
@kschiffer's proposal is sound; I'll cram this in for 3.14.1.

johanstokking, 2021-08-06:
Closed by #4489

closed 2021-08-06 by johanstokking