TheThingsNetwork/lorawan-stack #4397: Select Packet Forwarder while registering a Gateway on Console

i3mSuperMan, 2021-07-13:
#### Summary
Could we add a parameter `Packet-Forwarder` with a drop down of the packet-forwarders list (UDP, LBS) that are supported by The Things Stack to the gateway registration?
#### Why do we need this?
V2 - V3 similarities
#### What is already there? What do you see now?

- In V2, there is an option to select the legacy packet forwarder from the Console (refer to image).
[image]

- Currently, we have an option to download the `global_conf.json` file download option directly from the Console which is helpful to configure the radio settings of the UDP gateways.

#### What is missing? What do you want to see?

- Drop down of packet forwarders supported by The Things Stack while registering the Gateway
- We see that there is no support for downloading the `station.conf` file of Basics Station gateways. For Gateways configured/registered with Basics Station protocol, the download option should show `station.conf` file.

CC: @KrishnaIyer.
#### Environment
TTSC v3.13.3
#### How do you propose to implement this?
...
#### How do you propose to test this?
...
#### Can you do this yourself and submit a Pull Request?
No.

KrishnaIyer, 2021-07-13:
> Drop down of packet forwarders supported by The Things Stack while registering the Gateway

why is this needed?

> We see that there is no support for downloading the station.conf file of Basics Station gateways. For Gateways configured/registered with Basics Station protocol, the download option should show station.conf file.

`station.conf` is hardware specific and should be provided by the gateway manufacturer.

johanstokking, 2021-07-20:
> #### Why do we need this?
> V2 - V3 similarities

Can you elaborate why this is needed? We had a reason to have this in V2, because of identifiers, but this does not apply anymore.

i3mSuperMan, 2021-07-20:
Hi @KrishnaIyer & @johanstokking.

Our thought was about **having customized Gateway registration flow based on the selected packet forwarder**. However, we are not sure if it can fit across multiple distributions.

For example,  `Require authenticated connection` option might not be relevant for UDP gateways and few other fields while registering a gateway would be relevant for LBS only.

Also, it would be better to provide below information in overview page for the Users:
when using UDP Gateway.

- Gateway Server address
- Port
- Global configuration

Similarly, for LBS Gateways,

- CUPS/LNS Server address
- Ports
- Station configuration

johanstokking, 2021-07-20:
I see. Storing which packet forwarder is used will be implemented as part of supporting gateways in the device repository.

Then, we know what packet forwarder the gateway supports. When we know that, we can generate or show per-gateway relevant configuration and put a documentation link to the relevant page in the docs.

Internal ref https://github.com/TheThingsIndustries/lorawan-stack/issues/2018

closed 2021-07-20 by johanstokking