TheThingsNetwork/lorawan-stack #4409: Configure gateway visibility

johanstokking, 2021-07-14:
#### Summary

Configure gateway visibility.

#### Why do we need this?

For administrators to configure the visibility of gateways in Packet Broker.

#### What is already there? What do you see now?

Configuration of routing policies.

#### What is missing? What do you want to see?

Configuration of gateway visibilities. This works very similar to routing policies: there is a default, and there are specific visibility settings per Home Network.

#### How do you propose to implement this?

1. Extend API; `Pba` service
2. Implement CLI support
3. Implement Console support

#### How do you propose to test this?

Local test setup with Packet Broker staging or even production. Packet Broker already provides all the required functionality.

#### Can you do this yourself and submit a Pull Request?

Can do (1) and (2). Can review (3).

johanstokking, 2021-09-24:
Updated the original comment and re-assigning to @kschiffer as Console is left for closing the first phase.

Also adding `needs/triage` so we can plan the exact milestone.

kschiffer, 2022-03-02:
So per-network visibility settings are currently not implemented yet? Otherwise, we can already implement both in the Console.

johanstokking, 2022-03-02:
Let's keep it simple for now and set default gateway visibility only.

The rest is possible too but not needed for now. If we start with the default config checkboxes we're good and we can keep that if we expand functionality.

PR #5294 (2022-03-11): Gateway visibility packet broker

#### Summary

This PR implements Console support for gateway visibility on Packet Broker.
Closes #4409 

#### Changes

- Added a "Default gateway visibility" tab
- Gateway visibility form
- Cypress tests

[image]

#### Testing

Manually and cypress tests

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ] Documentati [... cut]
closed 2022-03-11 by ryaplots