TheThingsNetwork/lorawan-stack #4412: expand LoraWanOptions in Gateway general settings gives error

rcmcronny, 2021-07-15:
#### Summary
I want to change the gateway frequency plan. So i go to the gateway, choose general settings and expand the LoraWanOptions.
Then i change the frequency plan and hit save. I get a "An unknown error occurred. Please try again later.”
This seems to happen for all elements in this section, i can not change anything. I don't know, if that is caused bei the TTIG Gateway. 
Forumspost: https://www.thethingsnetwork.org/forum/t/ttig-on-v3-change-frequency-does-not-work/49805

#### Steps to Reproduce

1. Login to v3 console 
2. choose gateway
3. choose general settings
4. expand the LoraWanOptions.
5. change the frequency plan (or other things there)
6. hit save
7. there is a browser error message generated with red ""An unknown error occurred. Please try again later.” message.

#### What is already there? What do you see now?

Browser console shows in this time no errors or network requests, after some seconds there is a console message printed
Error: Promised response from onMessage listener went out of scope   ExtensionMessagingService.js:89:34
perhaps related, not sure

#### Environment
Linux / Windows testet with firefox and edge on win10 21h1
GatewayID: eu-de-we-me-1

johanstokking, 2021-07-15:
Refs #4411 

cc @kschiffer @bafonins

bafonins, 2021-07-15:
Hey @rcmcronny! Can you try updating you gateway LoRaWAN settings from the incognito tab of your browser or just using another browser? The error you provided seems to be related to extensions installed on your browser.

Just tested this functionality in the Console and I cannot reproduce this issue.

rcmcronny, 2021-07-15:
I have tried with incognito and with an additional system. On both the issue persists. I will try in the evenning at home again with some devices and report. Its not critical for me, so all good.

rcmcronny, 2021-07-18:
> 
> 
> I have tried with incognito and with an additional system. On both the issue persists. I will try in the evenning at home again with some devices and report. Its not critical for me, so all good.

Sorry for delay, tried it today at home 
firefox normal -> same error
Microsoft Edge (Version 91.0.864.70 (Offizielles Build) (64-Bit) -> same error 

**I took some screenshots of microsoft edge, as this is default with no addons or whatever (firefox have some)**
[image]

[image]

[image]

bafonins, 2021-07-21:
I can reproduce this on `3.13.3` deployment of TTN on all browsers. However, the issue seems to be fixed on staging. Will come back to this once the newest release is deployed and apply changes if necessary.

bafonins, 2021-07-27:
Tested on the `eu1` cluster on `v3.14.0` release. The issue seems to be resolved. @rcmcronny Please reopen if you still experience this issue

closed 2021-07-27 by bafonins

rcmcronny, 2021-07-27:
@bafonins works for me now, many Thanks !