TheThingsNetwork/lorawan-stack #4435: Add lorawan data to payload formatter

gruijter, 2021-07-21:
#### Summary

I would like to add information like SNR and RSSI to the payload formatter. But this type of info is not available in the formatter. See more background in this forum thread:

https://www.thethingsnetwork.org/forum/t/payload-formatter-v3-how-to-return-rssi-and-snr-to-a-thingspeak-channel/49946/

...

#### Why do we need this?

I would like to use webhook integration to other systems like Homey and ThingSpeak, and want to be able to forward information about the lorawan message like gateway, bandwidth  rssi, snr, etc etc
...

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?
All message info that is also available through the mqtt interface

...

#### Environment

...

#### How do you propose to implement this?
Make all info available so it can be returned as data in the formatter
...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

No

...

gruijter, 2021-08-05:
Bump. So what is going to happen with this request?

nejraselimovic, 2021-08-09:
> Bump. So what is going to happen with this request?

@gruijter this doesn't really have much to do with payload formatters, since RSSI and SNR data are part of the uplink message metadata, but are not present as part of the `frm_payload` (which is a payload that you decode using payload formatters to get the `decoded_payload`).

As the matter of fact, the whole uplink message (including SNR and RSSI metadata that are of your interest) is forwarded to every third party platform via both MQTT or HTTP integration, but it depends if it will be available for you on that platform on the way that the integrator intended the integration to work. Having this said, you can browse our documentation page to see some different approaches our integrators took, here are few examples:
- [Cayenne integration](https://www.thethingsindustries.com/docs/integrations/cloud-integrations/cayenne/) where SNR and RSSI are automatically extracted from an uplink message metadata and shown on the dashboard
- [ThingsBoard](https://www.thethingsindustries.com/docs/integrations/cloud-integrations/thingsboard/) integration where you can use their uplink data converter to extract SNR and RSSI from an uplink message metadata
- [ThingSpeak](https://www.thethingsindustries.com/docs/integrations/cloud-integrations/thingspeak/) integration where everything is predefined in a such way that you can't obtain any other info than what is in `decoded_payload` (which you define by decoding `frm_payload` on The Things Stack). 

Now, the only thing that we can do here is to contact integrators and forward this request to let them know that there are users interested in this feature, and I will for sure do that.

gruijter, 2021-08-13:
I also asked ThingSpeak (Matlab). See this link:
https://www.mathworks.com/matlabcentral/answers/883319-things-stack-ttn-v3-howto-write-lorawan-message-information-to-thingspeak-stream

NicolasMrad, 2022-03-29:
what's the progress here or can we close this issue?

nejraselimovic, 2022-03-30:
> what's the progress here or can we close this issue?

This is still in progress, but going slowly, so we can just move the milestone to Q2.

bodhibrata, 2022-04-05:
If this feature (storing RSSI and SNR value in ThingSpeak) is made available it will be of great help.

darkoj416, 2022-04-27:
> If this feature (storing RSSI and SNR value in ThingSpeak) is made available it will be of great help.

+1

NicolasMrad, 2022-12-12:
@nejraselimovic any updates here?

closed 2023-04-28 by NicolasMrad