TheThingsNetwork/lorawan-stack #4457: Documentation for "Adding Devices" is too simple and lacks essential information and details

lnlp, 2021-07-28:
#### Summary

Documentation for "[Adding Devices](https://www.thethingsindustries.com/docs/devices/adding-devices/)" is too simple and lacks essential information and details.

#### Why do we need this ?

The documentation is where we should be able to find all needed information,
however essential information and details are missing.

#### What is already there? What do you see now?

Instructions for how to add a device but the information is too simple. Essential information and details are missing.

Section "Manually Create End Device".

> This guide covers adding a device using OTAA (the most secure and preferred activation method) and LoRaWAN version MAC V1.0.2

- I only see "your manufacturer" mentioned. There seems to be no awareness of and no mentioning of DIY/custom devices.
- For Console ABP is missing. 
- For CLI ABP essential Advanced Mac settings are missing. 
- Other LoRaWAN versions are missing.

Don't shortcut the documentation with e.g. only a Console example for LoRaWAN 1.0.2 with OTAA.
For CLI examples show all parameters for a device not just a minimum subset.

People come here for the right information and shall not have to make any (impacting) guesses because the documentation lacks information.

> There are also advanced settings to configure Frame counter width and RX2 settings, but these should not be necessary for most devices

- Further explanation and/or additional information is missing here.
  Where else should users look for that information? They should be able to find it just here.

#### What is missing? What do you want to see?

See above.

#### How do you propose to document this?

Add the missing information.

#### Can you do this yourself and submit a Pull Request?

No.

kschiffer, 2021-07-30:
Thanks for filing this issue @lnlp! The documentation around end device registration has to be updated and extended indeed. Also since we now have a new manual registration form in the Console since 3.14.0.

@benolayinka do you think you can work on this? I can assist you with regards to the new Console form.

benolayinka, 2021-08-02:
Moved to https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/492

closed 2021-08-02 by benolayinka