TheThingsNetwork/lorawan-stack #4466: Put coding rate in band's data rate

johanstokking, 2021-07-29:
#### Summary

Declare coding rate in band's data rates

#### Why do we need this?

Unlike with LoRa, with LR-FHSS, the data rate is partly defined by the coding rate.

I think that this is needed for https://github.com/TheThingsNetwork/lorawan-stack/issues/3806

#### What is already there? What do you see now?

Gateway independently sets the PHY's LoRa coding rate on downlink.

With LR-FHSS this isn't possible; it depends on the data rate, but this is not declared in the band.

#### What is missing? What do you want to see?

The coding rate isn't defined in the band.

#### How do you propose to implement this?

I think we should declare the coding rate as part of the LR-FHSS data rate so that it can be get by index.

That is, add it as a field to `message LRFHSSDataRate` API.

We don't have to define the LoRa coding rate in `LoRaDataRate` just yet, let's not do that for now either.

#### How do you propose to test this?

Unit tests to get the full data rate by index and back

#### Can you do this yourself and submit a Pull Request?

Can review

adriansmares, 2022-08-18:
Implementation details:

1. Remove `LoRaCodingRate` from the `band.Band`, and from the instances themselves. You may already drop it from https://github.com/TheThingsNetwork/lorawan-stack/pull/5701 . 
https://github.com/TheThingsNetwork/lorawan-stack/blob/0b3d7b60d80abf0ffa799cbac68a14311616c363/pkg/band/band.go#L82-L83 
2. Add the coding rate to the LoRa data rate, in the same way LR-FHSS does this:
https://github.com/TheThingsNetwork/lorawan-stack/blob/0b3d7b60d80abf0ffa799cbac68a14311616c363/api/lorawan.proto#L397-L402
https://github.com/TheThingsNetwork/lorawan-stack/blob/0b3d7b60d80abf0ffa799cbac68a14311616c363/api/lorawan.proto#L410-L416
3. Update `band.makeLoRaDataRate` to also include the coding rate (after the bandwidth), and then update all of the usages of `makeLoRaDataRate` to include the `4/5` data rate (make this a constant). For the ISM 2.4 GHz band, make sure you use the `4/8LI` instead (constant again).
4. Lookup for references to this issue (`https://github.com/TheThingsNetwork/lorawan-stack/issues/4466`) and update the code to use the modulation coding rate.

ysmilda, 2022-08-18:
Would it be beneficial to remove the `LoRaCodingRate` from PR #5701 and cross contaminating these two issues? Or would it be better to wait for #5701 to be merged and then tackle this part of the issue?

adriansmares, 2022-08-18:
> Would it be beneficial to remove the `LoRaCodingRate` from PR #5701 and cross contaminating these two issues? Or would it be better to wait for #5701 to be merged and then tackle this part of the issue?

You may already remove it indeed. I (tried to) hint at this in point (1).

ysmilda, 2022-08-18:
This change causes the `TxSettings` message as defined here:
https://github.com/TheThingsNetwork/lorawan-stack/blob/0b3d7b60d80abf0ffa799cbac68a14311616c363/api/lorawan.proto#L432-L449
To have a double declaration of the `coding-rate` as it is bundled in the `DataRate` as well. What would the prefered sollution be here?

adriansmares, 2022-08-18:
`TxSettings` is a message that we try to keep in line with the Backend Interfaces specification. With that being said, I don't see the coding rate itself being part of the specification, and Packet Broker from what I see expects the coding rate to be in the LoRa Data Rate, so removal should be fine.

There is a note to be had here however - in some places (that do not reference this issue) the coding rate is obtained from `TxSettings`. In most of these places you will have the modulation available, so you can copy the copy the coding rate from there instead of `TxSettings`. The reverse may also happen - assignments to `TxSettings.CodingRate` may now have to go to the modulation instead (if it exists - for example for UDP packets it will come from the `CodR` field).

PR #5715 (2022-09-08): Put coding rate in band's data rate

#### Summary

Closes #4466 

#### Changes

- Define `coding_rate` in `LoRaDataRate` instead of `Band` definition
- Remove `coding_rate` from `TxSettings` as it is now defined in `DataRate`

#### Testing

UT

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [x] Documentation: Relevant documentation is added or update [... cut]
closed 2022-09-08 by ysmilda