TheThingsNetwork/lorawan-stack #4500: confusing error message if dev-eui is too short

christianwicke, 2021-08-07:
#### Summary

There is no check on the length of the dev-eui. 
Instead a technical error is shown: 
  
`error:pkg/types:invalid_length (invalid slice length)`

This error doesn't show which parameter is wrong.

#### Steps to Reproduce

run
`ttn-lw-cli end-devices create a b --net-id 000013 --join-eui 0000000000000000 --dev-eui 123456789012`

#### What do you see now?

`error:pkg/types:invalid_length (invalid slice length)`
#### What do you want to see instead?

`dev-eui is too short, must be 8 bytes (16 characters in hex)`

#### Environment

```
# ttn-lw-cli version
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.14.1
Build date:          2021-08-06T11:08:04Z
Git commit:          0be1360d5
Go version:          go1.16.7
OS/Arch:             linux/amd64
```

#### How do you propose to implement this?

The parameter length should be checked before running in a invalid slice length error

#### How do you propose to test this?

It can be tested easily (as I did to produced the bug)

#### Can you do this yourself and submit a Pull Request?

I haven't programmed in go yet. I think someone who knows the code can do this easily. I just want to create this bug because I spend some time figuring out what is wrong. I think this should be addressed so others don't fall in the same pit.

closed 2021-08-13 by htdvisser

christianwicke, 2021-08-14:
Thank you very much. You fixed this very fast. I wasn't expecting this.

htdvisser, 2021-08-16:
You're welcome!

Do keep in mind that this will land in the next release, so if you'd make the same mistake now, you'd still get the old error.

PR #4510 (2021-08-13): Wrap type unmarshaling errors

#### Summary

This pull request adds some error definitions so that we can better wrap type unmarshaling errors.

Instead of

```
error:pkg/types:invalid_length (invalid slice length)
```

you'll now see

```
error:cmd/ttn-lw-cli/commands:invalid_join_eui (invalid JoinEUI)
--- error:pkg/types:invalid_eui (invalid EUI)
--- error:pkg/types:invalid_length (invalid length: want 8 bytes, got 3)
```

Closes #4500

#### Changes

- Added some errInvalidTypeName errors
- Replaced bare `return err` with `return errInvalidTypeName.WithCause(err)`

#### Testing

CLI, existin [... cut]