TheThingsNetwork/lorawan-stack #4503: lora-basic-station and TTS disagree on the interpretation of EUI 80:fd46

jchonig, 2021-08-08:
#### Summary
Using mLinux 5.0.3d with lora-basic-station 2.0.5 with an EUI of 008000000000FD46 fails to connect with

```
2021-08-08 17:27:26.151 [AIO:XDEB] [4|WS] > {"router":"80::fd46"}
2021-08-08 17:27:26.380 [AIO:XDEB] [4|WS] < {"router":"::0","muxs":"::0","error":"Failed to fetch gateway: error:pkg/gatewayserver:gateway_eui_not_registered (gateway EUI `000000800000FD46` is not registered)"}
2021-08-08 17:27:26.380 [TCE:ERRO] Infos error: ::0 Failed to fetch gateway: error:pkg/gatewayserver:gateway_eui_not_registered (gateway EUI `000000800000FD46` is not registered)
```

lora-basic-station is parsing the EUI a *008000000000FD46* and as you can see above, passing it to TTS as *80::fd46* where it is being interpreted as *000000800000FD46*.  

If I change the EUI to on the console to *000000800000FD46* as a workaround, lora-basic-station will connect and function correctly.

...

#### Steps to Reproduce

1. Configure a gateway and TTS test instance with the given EUI
2. Try to connect

#### What do you see now?

```
2021-08-08 17:27:24.843 [SYS:INFO] Logging     : /var/log/basic-station.log (maxsize=10000000, rotate=14)
2021-08-08 17:27:24.892 [SYS:INFO] Station Ver : 2.0.5(mlinux/std) 2021-03-09 17:38:50
2021-08-08 17:27:24.892 [SYS:INFO] Package Ver : (null)
2021-08-08 17:27:24.892 [SYS:INFO] proto EUI   : 80::fd46       (station.conf)
2021-08-08 17:27:24.893 [SYS:INFO] prefix EUI  : ::0    (station.conf)
2021-08-08 17:27:24.893 [SYS:INFO] Station EUI : 80::fd46
2021-08-08 17:27:24.893 [SYS:INFO] Station home: ./     (builtin)
2021-08-08 17:27:24.893 [SYS:INFO] Station temp: /var/tmp/      (builtin)
2021-08-08 17:27:24.894 [SYS:WARN] Station in NO-CUPS mode
2021-08-08 17:27:25.111 [TCE:INFO] Starting TC engine
2021-08-08 17:27:25.176 [any:INFO] ./tc.trust:
cert. version     : 3
serial number     : 82:10:CF:B0:D2:40:E3:59:44:63:E0:BB:63:82:8B:00
issuer name       : C=US, O=Internet Security Research Group, CN=ISRG Root X1
subject name      : C=US, O=Internet Security Research Group, CN=ISRG Root X1
issued  on        : 2015-06-04 11:04:38
expires on        : 2035-06-04 11:04:38
signed using      : RSA with SHA-256
RSA key size      : 4096 bits
basic constraints : CA=true
key usage         : Key Cert Sign, CRL Sig2021-08-08 17:27:25.177 [AIO:INFO] tc has no cert configured - running server auth and client auth with token
2021-08-08 17:27:25.408 [AIO:XDEB] [4] ws_connecting state=1
2021-08-08 17:27:25.423 [TCE:INFO] Connecting to INFOS: wss://nam1.cloud.thethings.network:8887
2021-08-08 17:27:25.510 [AIO:XDEB] [4] ws_connecting state=1
2021-08-08 17:27:26.044 [AIO:XDEB] [4] ws_connecting state=1
2021-08-08 17:27:26.046 [AIO:XDEB] [4] ws_connecting state=2
2021-08-08 17:27:26.129 [AIO:XDEB] [4] ws_connecting state=3
2021-08-08 17:27:26.151 [AIO:XDEB] [4|WS] > {"router":"80::fd46"}
2021-08-08 17:27:26.380 [AIO:XDEB] [4|WS] < {"router":"::0","muxs":"::0","error":"Failed to fetch gateway: error:pkg/gatewayserver:gateway_eui_not_registered (gateway EUI `000000800000FD46` is not registered)"}
2021-08-08 17:27:26.380 [TCE:ERRO] Infos error: ::0 Failed to fetch gateway: error:pkg/gatewayserver:gateway_eui_not_registered (gateway EUI `000000800000FD46` is not registered)
2021-08-08 17:27:26.380 [AIO:DEBU] [4] ws_close reason=1000
2021-08-08 17:27:26.380 [AIO:ERRO] Recv failed: SSL - The peer notified us that the connection is going to be closed
2021-08-08 17:27:26.487 [AIO:DEBU] [4] WS connection shutdown...
2021-08-08 17:27:26.488 [TCE:INFO] INFOS reconnect backoff 0s (retry 0)
```
...

#### What do you want to see instead?

lora-basic-station connects and functions

...

#### Environment
mLinux 5.3.0d
lora-basic-station 2.0.5
nam1.cloud.thethings.network

station_conf section from station.conf:

```json
    "station_conf": {
        "TC_TIMEOUT": "2s",
        "euiprefix": "::0",
        "log_file": "/var/log/basic-station.log",
        "log_level": "XDEBUG",
        "log_rotate": 14,
        "log_size": 10000000,
        "routerid": "008000000000FD46"
    }
```

...

#### How do you propose to implement this?

#### How do you propose to test this?

#### Can you do this yourself and submit a Pull Request?

...

jchonig, 2021-08-08:
If I don't provide a *routerid* in station.conf, lora-basic-station will use the MAC address of *eth0*:

```
2021-08-08 21:17:09.195 [SYS:INFO] proto EUI   : 0:8:4a:2b2c    (/sys/class/net/eth0/address)
```

However TTS does not seem to interpret this the same way as lora-basic-station:

```
2021-08-08 21:17:10.071 [AIO:XDEB] [4|WS] > {"router":"0:8:4a:2b2c"}
2021-08-08 21:17:10.296 [AIO:XDEB] [4|WS] < {"router":"::0","muxs":"::0","error":"Failed to fetch gateway: error:pkg/gatewayserver:gateway_eui_not_registered (gateway EUI `00000008004A2B2C` is not registered)"}
```

It will only connect if I configure the EUI in the TTS GUI to match what TTS interprets.

Thanks.

Jeff

johanstokking, 2021-08-09:
@jchonig Thanks for reporting.

@KrishnaIyer I do see test failure with the ID:

```diff
diff --git a/pkg/basicstation/eui_test.go b/pkg/basicstation/eui_test.go
index 566a8294c3..0ae1ec7479 100644
--- a/pkg/basicstation/eui_test.go
+++ b/pkg/basicstation/eui_test.go
@@ -144,6 +144,16 @@ func TestUnmarshalEUI(t *testing.T) {
 			Input: `-12302426811387609088`,
 			OK:    false,
 		},
+		{
+			Input: `"80::fd46"`,
+			EUI64: types.EUI64{0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0xFD, 0x46},
+			OK:    true,
+		},
+		{
+			Input: `"0:8:4a:2b2c"`,
+			EUI64: types.EUI64{0x00, 0x80, 0x00, 0x00, 0x00, 0x4a, 0x2b, 0x2c},
+			OK:    true,
+		},
 	} {
 		t.Run(strconv.Itoa(i), func(t *testing.T) {
 			a := assertions.New(t)

```

However, I do see two distinct problems here.

First, `80::fd46` has three groups, while four are expected. The way we interpret this, is by filling in right-to-left, meaning that the fourth is the leftmost, which is empty, resulting in `0000 0080 0000 FD46`.

Second, `0:8:4a:2b2c` has four groups, and here you see clearly that the first block is `0000`, so it becomes correctly `0000 0080 004A 2B2C`.

So, I don't think there's anything we do wrong on our side.

@reissjason what do you think?

jchonig, 2021-08-09:
@johanstokking 

Re: `80::fd46`

Basic Station ID6 addresses are an 64-bit form of an IPv6 address.   That is briefly mentioned in the [Basic Station documentation](https://doc.sm.tc/station/glossary.html) .

According the the IPv6 rules, this should be `0080 0000 0000 FD46`.

An ellipsis (::) is *not* just one group of zero.   An ellipsis needs to be replaced with one or more groups of zeros until the left-most group is non-zero.  Unless there is a leading ellipsis in which case the remaining groups are right aligned.

There is an example in the [Basic Station documentation](https://doc.sm.tc/station/glossary.html) that makes this clear: `f::1 = 00-0f-00-00-00-00-00-01`.  A test for that would be:

```go
		{
			Input: `"f::1"`,
			EUI64: types.EUI64{0x00, 0x0f, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01},
			OK:    true,
		},
```

jchonig, 2021-08-09:
Re: `0:8:4a:2b2c`

The MAC address that the Basic Station is trying to represent as `0:8:4a:2b2c` is `00:08:00:4a:2b:2c`.   That would be a test of:

```go
		{
			Input: `"0:8:4a:2b2c"`,
			EUI64: types.EUI64{0x00, 0x00, 0x00, 0x08, 0x00, 0x4a, 0x2b, 0x2c},
			OK:    true,
		},
```
The log of it trying to connect shows:

```
2021-08-09 22:06:43.543 [AIO:XDEB] [4|WS] > {"router":"0:8:4a:2b2c"}
2021-08-09 22:06:43.770 [AIO:XDEB] [4|WS] < {"router":"::0","muxs":"::0","error":"Failed to fetch gateway: error:pkg/gatewayserver:gateway_eui_not_registered (gateway EUI `00000008004A2B2C` is not registered)"}
2021-08-09 22:06:43.770 [TCE:ERRO] Infos error: ::0 Failed to fetch gateway: error:pkg/gatewayserver:gateway_eui_not_registered (gateway EUI `00000008004A2B2C` is not registered)
```

Even though I updated the EUI in the GUI, the CLI still returns the EUI below.  Which means that this is a different bug, updating the EUI in the GUI claims success but does not update the database:

```json
  "ids": {
    "gateway_id": "ttn-ith-test2",
    "eui": "008000000000FD46"
  },
```

jchonig, 2021-08-09:
Here is the [code](https://github.com/lorabasics/basicstation/blob/master/src/uj.c#L874-L896) that encodes the ID6 on the Basic Station side.

reissjason, 2021-08-10:
I agree with @jchonig on the parsing of double colons according to the Basic Station example.
https://github.com/TheThingsNetwork/lorawan-stack/issues/4503#issuecomment-895586247

The first test case @johanstokking posted should pass and the second should fail.
https://github.com/TheThingsNetwork/lorawan-stack/issues/4503#issuecomment-895532091

https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.14/pkg/basicstation/eui.go
See the id6Pattern.

Here is an ipv6 example
https://ipv6-literal.com/?ipv6=80%3A%3Afd46

This regex match does not look correct.
> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):)?([a-f0-9]{0,4}):([a-f0-9]{0,4}):([a-f0-9]{0,4})$/.exec("80::fd46")
[ '80::fd46',
  undefined,
  undefined,
  '80',
  '',
  'fd46',
  index: 0,
  input: '80::fd46' ]

0x0080 should be the value of the first group.
0x00 and 0x00 the second and third
0xfd46 the last

These two examples with prefixes should be equivalent.
> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):)?([a-f0-9]{0,4}):([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-80::fd46")
[ 'router-80::fd46',
  'router',
  undefined,
  '80',
  undefined,
  'fd46',
  index: 0,
  input: 'router-80::fd46' ]
> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):)?([a-f0-9]{0,4}):([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-80:00:00:fd46")
[ 'router-80:00:00:fd46',
  'router',
  '80',
  '00',
  '00',
  'fd46',
  index: 0,
  input: 'router-80:00:00:fd46' ]

I would add test cases of "f::1" = "000f000000000001, "::1" = "0000000000000001 and "1::" = "0001000000000000"

This regex appears to work.

> /^(?:([a-z]+)-)?(?:([a-f0-9]{1,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-f::1")
[ 'router-f::1',
  'router',
  'f',
  undefined,
  undefined,
  '1',
  index: 0,
  input: 'router-f::1' ]
> /^(?:([a-z]+)-)?(?:([a-f0-9]{1,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-0:8:4a:2b2c")
[ 'router-0:8:4a:2b2c',
  'router',
  '0',
  '8',
  '4a',
  '2b2c',
  index: 0,
  input: 'router-0:8:4a:2b2c' ]
> /^(?:([a-z]+)-)?(?:([a-f0-9]{1,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-80:fd46")
null
> /^(?:([a-z]+)-)?(?:([a-f0-9]{1,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-80::fd46")
[ 'router-80::fd46',
  'router',
  '80',
  undefined,
  undefined,
  'fd46',
  index: 0,
  input: 'router-80::fd46' ]

jchonig, 2021-08-10:
@ reissjason

Your regexp works, except for the `::X` cases and `::X:Y` cases, with or without the `name-` prefix.

jchonig, 2021-08-10:
@reissjason 

Your regex doesn't match `::X`, `::X:Y`, with our w/o the `name-` prefix.

reissjason, 2021-08-10:
Looks like I had a mistake in the first hex match where I changed the 0,4 to 1,4.

> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("::1:2")
[ '::1:2',
  undefined,
  '',
  undefined,
  '1',
  '2',
  index: 0,
  input: '::1:2' ]
> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-::1:2")
[ 'router-::1:2',
  'router',
  '',
  undefined,
  '1',
  '2',
  index: 0,
  input: 'router-::1:2' ]
> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-1::")
[ 'router-1::',
  'router',
  '1',
  undefined,
  undefined,
  '',
  index: 0,
  input: 'router-1::' ]

It does not rule out some invalid cases though, neither did the original.

> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-1:3:4")
[ 'router-1:3:4',
  'router',
  '1',
  undefined,
  '3',
  '4',
  index: 0,
  input: 'router-1:3:4' ]
> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-1:")
null
> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-1:1")
null
> /^(?:([a-z]+)-)?(?:([a-f0-9]{0,4}):?)([a-f0-9]{0,4})?:([a-f0-9]{0,4})?:([a-f0-9]{0,4})$/.exec("router-1:1:")
[ 'router-1:1:',
  'router',
  '1',
  undefined,
  '1',
  '',
  index: 0,
  input: 'router-1:1:' ]

closed 2021-08-10 by johanstokking