TheThingsNetwork/lorawan-stack #4518: Expose `band` pkg via API to enable sourcing FP constraints

kschiffer, 2021-08-11:
#### Summary
We need to come up with a way to expose the restrictions of frequency plans with regards to MAC/PHY versions.

#### Why do we need this?
E.g. the Australian FPs do not support MAC/PHY 1.0.0. However, the Console still allows setting this combination because the frontend is unaware of such restrictions. This leads to frustration when users fill out the end device registration form and hitting this limitation only upon submitting.

[image]

#### What is already there? What do you see now?
- End device registration form

#### What is missing? What do you want to see?
- API to query from the Console to get the `band -> regional parameters version -> available` mapping

#### How do you propose to implement this?
Probably trivial. cc @adriansmares 

#### How do you propose to test this?
Manual testing and updating e2es.

#### Can you do this yourself and submit a Pull Request?
Need help from @adriansmares (or whoever else feels confident with this).

kschiffer, 2021-08-11:
Currently blocking #4519

KrishnaIyer, 2021-08-25:
Moving this due to time constraints. Will pick it up early next milestone.

KrishnaIyer, 2021-09-08:
This is actually related to https://github.com/TheThingsNetwork/lorawan-stack/pull/4591/files. We can add an RPC to the cluster package to fetch supported bands.

adriansmares, 2021-09-08:
> This is actually related to https://github.com/TheThingsNetwork/lorawan-stack/pull/4591/files. We can add an RPC to the cluster package to fetch supported bands.

You can already build this using `band.All` - the usage won't change _too much_ after #4591 is done.

PR #4605 (2021-09-13): RPC to fetch supported PHY versions for Band ID

#### Summary

RPC to fetch supported PHY versions for Band ID. Closes #4518 

#### Changes

- Add a new `GetSupportedPhyVersions` to the configuration server.
- Implement RPC and add tests.

#### Testing

UT

##### Regressions

NA

#### Notes for Reviewers

The implementation can change a little after https://github.com/TheThingsNetwork/lorawan-stack/pull/4591 but the API will not.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existi [... cut]
closed 2021-09-13 by KrishnaIyer