TheThingsNetwork/lorawan-stack #4526: Disable webhook configuration after too many failures

htdvisser, 2021-08-13:
#### Summary

I think we should build functionality into the AS Webhooks that disables webhook configurations after too many failures.

#### What is already there? What do you see now?

The AS is wasting a lot of time attempting webhooks that never succeed.

#### What is missing? What do you want to see?

Disable webhook configurations after a (configurable) number of failures.

#### How do you propose to implement this?

- Some configuration for number of failures, types of failures, when to reset the counters, etc.
- Add fields in the webhook configuration message for keeping track of the number of failures, maybe a field for one or multiple error messages.
- Keep track of the number of failures for a webhook configuration.
- If this number crosses a certain threshold, don't even bother trying.
- Display a warning in the Console if there's a high number of failures.
- Allow users to re-enable the webhook configuration.
- Maybe re-enable the webhook configuration automatically?

NicolasMrad, 2021-08-17:
https://github.com/TheThingsIndustries/lorawan-stack/issues/2440

adriansmares, 2021-12-08:
Now that the backend (and a very primitive frontend) is available as part of https://github.com/TheThingsNetwork/lorawan-stack/pull/4913, I'd like to point out how the fields work.

The webhook entity now has a `health_status` field. By default, it is `null`/`undefined`, indicating that the health status is unknown since an HTTP request attempt has not been made yet.

If the HTTP request succeeds, inside `health_status` an empty `healthy` object will become available. This signifies that the webhook 'works', and that this is basically the happy flow.

If the HTTP request fails, inside `health_status` an object called `unhealthy` will become available. This signifies that the webhook has recently encountered a failure, and the contents of the object contain details about this failure.

```json
{
	"health_status": {
		"unhealthy": {
			"failed_attempts": "1",
			"last_failed_attempt_at": "2021-12-07T21:00:19.364658787Z",
			"last_failed_attempt_details": {
				"namespace": "pkg/applicationserver/io/web",
				"name": "request",
				"message_format": "request",
				"correlation_id": "930f922e06574d3186e393bd3168a112",
				"code": 14,
				"details": [
					{
						"@type": "type.googleapis.com/google.protobuf.Struct",
						"value": {
							"body": "<!DOCTYPE html>\n<html lang=en>\n  <meta charset=utf-8>\n  <meta name=viewport content=\"initial-scale=1, minimum-scale=1, width=device-width\">\n  <title>Error 405 (Method Not Allowed)!!1</title>\n  <style>\n    *{margin:0;padding:0}html,code{font:15px/22px arial,sans-serif}html{background:#fff;color:#222;padding:15px}body{margin:7% auto 0;max-width:390px;min-height:180px;padding:30px 0 15px}* > body{background:url(//www.google.com/images/errors/robot.png) 100% 5px no-repeat;padding-right:205px}p{margin:11px 0 22px;overflow:hidden}ins{color:#777;text-decoration:none}a img{border:0}@media screen and (max-width:772px){body{background:none;margin-top:0;max-width:none;padding-right:0}}#logo{background:url(//www.google.com/images/branding/googlelogo/1x/googlelogo_color_150x54dp.png) no-repeat;margin-left:-5px}@media only screen and (min-resolution:192dpi){#logo{background:url(//www.google.com/images/branding/googlelogo/2x/googlelogo_color_150x54dp.png) no-repeat 0% 0%/100% 100%;-moz-border-image:url(//www.google.com/imag",
							"status_code": 405,
							"url": "https://google.com",
							"webhook_id": "wh1"
						}
					}
				]
			}
		}
	}
}
```

`failed_attempts` is pretty much what it says - the number of successive failures encountered while the webhook is unhealthy.
`last_failed_attempt_at` is again self explanatory - it is the timestamp of the last HTTP request attempt.
`last_failed_attempt_details` is an `ErrorDetails` object that contains some debugging information about the failure - under `details` you can see the `status_code` of the HTTP request, the URL to which the request was sent, and a fragment of the body of the response.

The Application Server configuration is also available via API. This is available in the frontend, but needs to be backported from Enterprise - see [this](https://github.com/TheThingsIndustries/lorawan-stack/blob/0906e8b33239cd4eff2769151fbf22e352a97835/pkg/webui/console/store/middleware/logics/application-server.js)

The configuration object looks as follows:
```json
{
	"configuration": {
		"pubsub": {
			"providers": {
				"mqtt": "WARNING",
				"nats": "WARNING"
			}
		},
		"webhooks": {
			"unhealthy_attempts_threshold": "16",
			"unhealthy_retry_interval": "86400s"
		}
	}
}
```

The frontend may provide quite some contextual help.  For example, if `failed_attempts` is smaller than `unhealthy_attempts_threshold`, the Console may convey that the next HTTP request will be attempted, but if it fails the number of attempts will increase.

If `failed_attempts` is greater or equal to `unhealthy_attempts_threshold`, then the `last_failed_attempt_at` can be used to relay what will happen to the next request. If `last_failed_attempt_at` + `unhealthy_retry_interval` is a moment in the past, the request will be executed, but if it is a moment in the future, the request will be dropped (no attempt will be made).

It can also be the case that `unhealthy_attempts_threshold` is `0`. This means that the Application Server will be doing strictly book keeping - the number of attempts will increase, the timestamp will be updated, the last error will be stored, but the requests will always be made - basically we're only monitoring the webhook. In such cases I would recommend that the Console conveys a friendly message that tells the users to get their webhook fixed asap or at least delete it if it's not used.

The Console may also allow the user to reset the health status of a webhook, by setting the `health_status` field to `null`.

closed 2021-12-15 by adriansmares