TheThingsNetwork/lorawan-stack #4527: Require successful webhook transmission before enabling the configuration

htdvisser, 2021-08-13:
#### Summary

When setting up a webhook configuration, we should verify that we can send webhooks to it before enabling the configuration.

#### What is already there? What do you see now?

Right now we have obviously incorrect webhook configurations.

#### What is missing? What do you want to see?

When creating a webhook, the AS should attempt to send a test payload, and only enable the webhook configuration if that succeeds.

#### How do you propose to implement this?

- Add an `enabled` boolean to the webhook configuration
- When a user changes `enabled` from `false` to `true`, send a test payload, and reject the change if it does not succeed.

jpmeijers, 2021-09-20:
This would be a useful feature, but remote servers might not like these test POSTs.

Another option to help users configure a webhook correctly is to show the last response from the remote server. The Sigfox backend does this, along with the full json body and headers. This allows one to replay the same post using a tool like Postman, at a user determined time, while running your server code in a debugger.

Example:
[image]

dajtxx, 2022-03-09:
> This would be a useful feature, but remote servers might not like these test POSTs.

That was my immediate thought too. Messages sent from the end device messaging page have a simulated: true flag in them which end points should take notice of. So perhaps that can be used here too.

KrishnaIyer, 2025-04-30:
A webhook is first placed in the `pending` state upon createion and we have webhook health checks that fail if a webhook fails. This is covered.

closed 2025-04-30 by KrishnaIyer