TheThingsNetwork/lorawan-stack #4530: Random CLI config options allowed

nejraselimovic, 2021-08-13:
#### Summary
When I add a random config option (`z-random-config-option` in the screenshot below) to `tti-lw-cli.yml file and list the configuration with `tti-lw-cli config`, everything seems normal and no error is reported. This leads to a confusion that this random config option is legit, but you soon realize this config option doesn't work.

Thanks to this I wasn't aware that `tenant-admin-key` wasn't a valid config option, so I couldn't understand why running `tti-lw-cli tenant list --tenant-admin-key <key-value>` worked, but setting `tenant-admin-key` in `tti-lw-cli.yml` and running `tti-lw-cli tenant list` didn't work.
[image]

#### What do you see now?
No indication of a bad config option.

#### What do you want to see instead?
Indication of a bad config option.

#### Environment
Enterprise CLI (didn't check for Open Source)

#### Can you do this yourself and submit a Pull Request?
@neoaggelos

neoaggelos, 2021-08-16:
Some more context, this is coming from `(*viper.Viper).AllKeys() []string` (in https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.14/pkg/config/config.go#L78), which returns all keys read from config files.

Possible solutions would be to:

- Stop using `AllKeys()` for iterating over config options, or use reflection on the config struct to check if a set key is actually valid config. This is mostly for printing config in `--env` format (for `yaml`, we could just marshal the config struct itself).
- Do nothing, document that invalid config keys will still show up.

I suppose it depends on how important we consider this issue to be. Glancing over Viper I didn't see any related options and/or GitHub issues, so I doubt it comes up too often in other projects. Unless I missed something too obvious.

cc @htdvisser

htdvisser, 2021-08-16:
I would rather not touch the config package too much right now; it's way too easy to break something, and since config is covered by our compatibility commitment we should really avoid that. On the other hand, I'm generally not too happy with the quality of the config package, and I think that it should probably be re-implemented at some point in the future. But I don't think now is the time for that.

Can't we catch these in the config command? All of these config options should be registered as flags, so maybe we can ignore keys for which we don't have flags?

Try adding

```go
if cmd.Flags().Lookup(key) == nil {
	continue
}
```

below

https://github.com/TheThingsNetwork/lorawan-stack/blob/2e17d00aaa9f186600425a8dbe73a602657db2cc/cmd/internal/commands/config.go#L50

PR #4535 (2021-08-17): Do not print invalid configuration keys

#### Summary

Closes #4530 

#### Changes

- When iterating over configuration keys, ignore the ones without any flags set.

#### Testing

Test with invalid configuration and `ttn-lw-cli config` and `ttn-lw-stack config` commands. Also made sure that no existing config is removed from the output.

##### Regressions

#### Notes for Reviewers

Change was suggested in https://github.com/TheThingsNetwork/lorawan-stack/issues/4530#issuecomment-899526439. Included an inline comment for future reference.

#### Checklist

- [X] Scope: The referenced issue is ad [... cut]
closed 2021-08-17 by neoaggelos