TheThingsNetwork/lorawan-stack #4536: Outdated documentation on official website

zgldh, 2021-08-17:
#### Summary

No need the flag ` --logout-redirect-uri` when creating the OAuth client for console.
...

#### Why do we need this ?

Please check this page https://www.thethingsindustries.com/docs/getting-started/installation/running-the-stack/
The command of this paragraph: Afterwards, the same needs to be done for the Console.
...

#### What is already there? What do you see now?

The flag `  --logout-redirect-uri` is not required actually.
stack will return `unknown flag: --logout-redirect-uri`
...

#### What is missing? What do you want to see?

Please remove flags `--logout-redirect-uri`
...

#### How do you propose to document this?

The flag `  --logout-redirect-uri` is not required actually.
...

#### Can you do this yourself and submit a Pull Request?

no
...

htdvisser, 2021-08-17:
The documentation is correct, when registering the Console client, we do need the `--logout-redirect-uri` flag. @michalborkowski96 can you try and reproduce this `unknown flag: --logout-redirect-uri` error?

zgldh, 2021-08-17:
[image]

michalborkowski96, 2021-08-24:
@zgldh I cannot reproduce this. Is it possible that you're using an old image? As in, you fetched the `latest` image long ago, before `logout-redirect-uri` was introduced [in this commit](https://github.com/TheThingsNetwork/lorawan-stack/commit/bcbb79bf5783db87117935004e4c97680cecd1bb)? Try:

1. Cleaning cache so that Docker fetches the `latest` image again.
2. Specifying `3.14.1` image, so that you're sure you're using the latest version.
3. Running `docker-compose run --rm stack version`, it should display `3.14.1`

zgldh, 2021-08-25:
I think I got the old version: 3.7.0 
[image]

I will do the updating.
Thank you very much!

closed 2021-08-25 by zgldh