TheThingsNetwork/lorawan-stack #4546: Google Cloud integration

nejraselimovic, 2021-08-18:
#### Summary
I got a question from The Things Summer Academy student whether a Google Cloud integration is possible? I have no idea how that works for IoT or LoRaWAN, @adriansmares perhaps you can share your thoughts?

#### Why do we need this?
...

#### What is already there? What do you see now?
...

#### What is missing? What do you want to see?
...

#### Environment
...

#### How do you propose to implement this?
...

#### How do you propose to test this?
...

#### Can you do this yourself and submit a Pull Request?
...

NicolasMrad, 2021-08-24:
at the moment Google Cloud integration is not possible. Will keep this issue opened for community votes and future reference.

KrishnaIyer, 2025-04-30:
Let's not spend time on this for now. We'll revisit it if necessary.

closed 2025-04-30 by KrishnaIyer