TheThingsNetwork/lorawan-stack #4551: Unable to register a device with "duplicate_identifiers" error

romach, 2021-08-20:
#### Summary

When I try to register a device, i got `duplicate_identifiers` error

#### Steps to Reproduce

1. Go to the application in the Things Network console
2. Click on "Add end device"
3. Select "Manually" tab
4. Fill device information
5. Click on "Register end device" button

#### What do you see now?

See error: Duplicate identifiers
```
{
  "code": 6,
  "message": "error:pkg/networkserver/redis:duplicate_identifiers (duplicate identifiers)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/networkserver/redis",
      "name": "duplicate_identifiers",
      "message_format": "duplicate identifiers",
      "correlation_id": "5c5bbb4a61b3414196ee2cdcf70fa37f",
      "code": 6
    }
  ],
  "request_details": {
    "url": "/ns/applications/4a86cc0f-98ec-4065-8e3c-d2f83cabd68a/devices/eui-647fda0000007fe0",
    "method": "put",
    "request_id": "92023e81-ac42-4708-84dd-ac554c55e1ef",
    "stack_component": "ns"
  }
}
```
#### What do you want to see instead?

List of devices

#### Environment

...

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

...

NicolasMrad, 2021-08-24:
To be able to troubleshoot this issue correctly, we need more background information from your side.
Did you register this device previously and deleted it? If so, did you receive any errors upon trying to delete the device?
And can you please inform us about the environment that you're using?

adriansmares, 2021-09-14:
I've been unable to reproduce this, as I cannot find the original registration.

The `duplicate_identifiers` error occurs when the `(device EUI: join EUI)` pair is already registered in a routing cluster (and this applies on a per component basis - the originally reported error was in the Network Server). 

I cannot find any reservation for the `647fda0000007fe0` EUI in any routing cluster on TTN (`eu1`, `nam1`, `au1`), so I don't see how this problem occurs. 

Any chance you could provide a minimal reproduction @romach ?

romach, 2021-09-15:
@adriansmares 
I can reproduce this bug using the following steps:
- register device manually with settings:
  - LoRaWAN version - MAC V1.0.3
  - Regional Parameters version - filled automatically/leave unchanged - PHY V1.0.3 REV A
  - Frequency plan - Europe 863-870 MHz (SF9 for RX2 - recommended)
  - DevEUI - 647FDA0000007FE0
  - AppEUI - 647FDA8010000100
  - AppKey - 92F7DEEA31F9A8D331362127B4B4FDDA

During the device registration four requests were sent by frontend:

1) Status code - 200
```
POST https://eu1.cloud.thethings.network/api/v3/applications/agro-bug-reproduce/devices

Body:

{"end_device":{"ids":{"dev_eui":"647FDA0000007FE0","join_eui":"647FDA8010000100","device_id":"eui-647fda0000007fe0"},"network_server_address":"eu1.cloud.thethings.network","application_server_address":"eu1.cloud.thethings.network","join_server_address":"eu1.cloud.thethings.network"},"field_mask":{"paths":["network_server_address","application_server_address","join_server_address"]}}

Response:

{"ids":{"device_id":"eui-647fda0000007fe0","application_ids":{"application_id":"agro-bug-reproduce"},"dev_eui":"647FDA0000007FE0","join_eui":"647FDA8010000100"},"created_at":"2021-09-15T13:18:33.461Z","updated_at":"2021-09-15T13:18:33.461Z","version_ids":{},"network_server_address":"eu1.cloud.thethings.network","application_server_address":"eu1.cloud.thethings.network","join_server_address":"eu1.cloud.thethings.network"}
``` 

2) Status code - 409
```
PUT https://eu1.cloud.thethings.network/api/v3/ns/applications/agro-bug-reproduce/devices/eui-647fda0000007fe0

Body:

{"end_device":{"frequency_plan_id":"EU_863_870_TTN","lorawan_phy_version":"PHY_V1_0_3_REV_A","multicast":false,"supports_join":true,"lorawan_version":"MAC_V1_0_3","ids":{"dev_eui":"647FDA0000007FE0","join_eui":"647FDA8010000100","device_id":"eui-647fda0000007fe0"},"supports_class_c":false,"supports_class_b":false,"mac_settings":{"rx2_data_rate_index":0,"rx2_frequency":869525000}},"field_mask":{"paths":["frequency_plan_id","lorawan_phy_version","multicast","supports_join","lorawan_version","ids.dev_eui","ids.join_eui","ids.device_id","supports_class_c","supports_class_b","mac_settings.rx2_data_rate_index","mac_settings.rx2_frequency"]}}

Response:

{"code":6,"message":"error:pkg/networkserver/redis:duplicate_identifiers (duplicate identifiers)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/networkserver/redis","name":"duplicate_identifiers","message_format":"duplicate identifiers","correlation_id":"d573362fc557448bbffabd185324adae","code":6}]}
```

3) Status code - 409

```
PUT https://eu1.cloud.thethings.network/api/v3/as/applications/agro-bug-reproduce/devices/eui-647fda0000007fe0

Body:

{"end_device":{"ids":{"dev_eui":"647FDA0000007FE0","join_eui":"647FDA8010000100","device_id":"eui-647fda0000007fe0"}},"field_mask":{"paths":["ids.dev_eui","ids.join_eui","ids.device_id"]}}

Response:

{"code":6,"message":"error:pkg/applicationserver/redis:duplicate_identifiers (duplicate identifiers)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/applicationserver/redis","name":"duplicate_identifiers","message_format":"duplicate identifiers","correlation_id":"65afc787e48740e5bd100eee8d69a76c","code":6}]}
```

4) Status code - 409

```
PUT https://eu1.cloud.thethings.network/api/v3/js/applications/agro-bug-reproduce/devices/eui-647fda0000007fe0

Body:

{"end_device":{"ids":{"dev_eui":"647FDA0000007FE0","join_eui":"647FDA8010000100","device_id":"eui-647fda0000007fe0"},"network_server_address":"eu1.cloud.thethings.network","application_server_address":"eu1.cloud.thethings.network","root_keys":{"app_key":{"key":"92F7DEEA31F9A8D331362127B4B4FDDA"}}},"field_mask":{"paths":["network_server_address","application_server_address","ids.dev_eui","ids.join_eui","ids.device_id","root_keys.app_key.key"]}}

Response:

{"code":6,"message":"error:pkg/joinserver/redis:duplicate_identifiers (duplicate identifiers)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/joinserver/redis","name":"duplicate_identifiers","message_format":"duplicate identifiers","correlation_id":"1bff73a0fc034fada8a6d6df670e57e7","code":6}]}
```

After that DELETE request was sent:

```
DELETE https://eu1.cloud.thethings.network/api/v3/applications/agro-bug-reproduce/devices/eui-647fda0000007fe0
```

adriansmares, 2021-09-15:
Ok so I finally found out what happened. The devices where registered in the IS/JS/AS using device ID `253486cc-a6db-43a2-a8c1-58dd0a2fceb3`. 

The Console (the frontend) assumes that the same device ID is used everywhere, and issued the `DELETE` requests under this assumption. But in your scenario, the IS device ID was `eui-647fda0000007fe0` while in the routing components the device ID was `253486cc-a6db-43a2-a8c1-58dd0a2fceb3`. But since you manually created the device, using the wrong device ID, this assumption was broken.

The solution was to issue the `DELETE` requests with the `253486cc-a6db-43a2-a8c1-58dd0a2fceb3` device ID (i.e. `DELETE https://eu1.cloud.thethings.network/api/v3/as/applications/4a86cc0f-98ec-4065-8e3c-d2f83cabd68a/devices/253486cc-a6db-43a2-a8c1-58dd0a2fceb3`).

Could you please confirm that you can reuse the EUI now ?

romach, 2021-09-15:
@adriansmares I can reuse EUI now - device is added successfully

closed 2021-09-15 by adriansmares

romach, 2021-10-13:
Hi @adriansmares I have similar problem in Australia region with device which has EUI 647FDA00000034F7.

Could you please delete it from IS/JS/AS ?